LiberTEM/LiberTEM #142: Automate the release process

sk1p, 2018-10-30:
Right now, there is some manual work involved with releasing LiberTEM, like building a wheel and sdist, uploading that to both PyPi and github, which can only happen after the tests have finished running etc...

What I'd like to have is a more automated process, that only requires tagging the release in source control. Basically how it is now done with the AppImage, but also for wheel and sdist. If a commit is tagged with a version tag, the following should happen:

1) tests run as usual
2) if tests complete successfully, build wheel & sdist
3) upload to github (how? somehow via API?) and pypi (with twine)

This needs to be integrated with the AppImage stuff, because the AppImage uploader creates the GitHub release, where we merely want to _add_ the wheel. Also, uploading to github and pypi should only happen if the AppImage builds successfully.

For 1st-class Windows support, we should think about moving the Windows CI also to travis, to fail the release process if the tests fail on Windows. Note: Windows support is still quite new on travis.

uellue, 2018-11-05:
We tested zenodo.org integration. It turns out that we should integrate the publication on zenodo.org conditionally in the Continuous Integration pipeline. If you activate the automation process on zenodo.or, every single push on the repository will create a release for continuous which is published on zenodo.org, and tagged releases will fail because zenodo.org recognizes them as duplicates. We should only publish proper releases. According to zenodo.org support, this is done by integrating this via their API. That way we control what is published and what isn't.

sk1p, 2018-11-27:
Wheel and sdist are now built in continuous integration and uploaded with the appimage upload tool. What is now still missing is conditional upload to pypi: we only want release candidates and full releases to go up on pypi. I guess easiest would be to make a separate shell script that runs directly after the appimage uploading. That script can check the version and upload to twine using credentials passed [from environment variables](https://twine.readthedocs.io/en/latest/#environment-variables). I think a check if "dev" is in the version string should be sufficient for now, but I need to check my notes on that.

Hmm. The same script could, if the release is not a release candidate, initiate the zenodo publishing process via their API (probably as a small Python script). Zenodo provides a [sandbox](http://developers.zenodo.org/#testing) where we can test our script. Maybe we could upload "rc" and "dev" versions on the sandbox, that way we can test the whole process.

sk1p, 2019-02-05:
For the PyPi part, we can have a look at [the .travis.yml of django channels](https://github.com/django/channels/blob/master/.travis.yml), for example. Relevant is the `deploy: [...] provider: pypi` part. This should be easily integrated, with a condition that matches the release tags (something that matches `vX.Y.Z(rcA)?`)

sk1p, 2019-08-14:
Still missing then is the parent **deposition** id, which should be set in the [travis settings](https://travis-ci.org/LiberTEM/LiberTEM/settings) as environment variable `LT_RELEASE_UPLOAD_ZENODO_PARENT` (can apparently only be found via the API, or via non-public edit / new version pages). Also, a new personal access token needs to be added as env var `LT_RELEASE_UPLOAD_ZENODO_TOKEN`. As there are [no write permissions for multiple users](https://github.com/zenodo/zenodo/issues/151) yet in zenodo, I think @uellue has to take care of this (the current access token is bound to my user, which doesn't have access to the record...)

sk1p, 2019-08-16:
Hm, zenodo upload to sandbox currently also fails, so I'm disabling it for now (see [the build log](https://travis-ci.org/LiberTEM/LiberTEM/jobs/571903038)).

Both Github releases and PyPi uploads work fine for release candidates, see https://github.com/LiberTEM/LiberTEM/releases/tag/v0.2.0rc7 and https://test.pypi.org/project/libertem/

uellue, 2019-08-27:
I've updated parent and token for both sandbox and production to my account. How do we test?

sk1p, 2019-08-27:
I just cut [a v0.2.0rc8 release](https://github.com/LiberTEM/LiberTEM/releases/tag/v0.2.0rc8), let's see if the zenodo upload runs through without problems.

sk1p, 2019-08-27:
Release job logs: https://travis-ci.org/LiberTEM/LiberTEM/jobs/577408484

sk1p, 2019-08-27:
Ugh, forgot to actually re-enable zenodo upload, let's try again...

sk1p, 2019-08-27:
Uploads to zenodo sandbox, github and [test.pypi.org](https://test.pypi.org/project/libertem/) look ok, still need to do some qa, after writing docs I'd say...

uellue, 2019-09-24:
It looks like we can leave the last item as an improvement in case that becomes an issue with future releases? Closing this for now since automated releases seem to work smoothly, at least for release candidates. :-)

closed 2019-09-24 by uellue