LiberTEM/LiberTEM #144: Implement big-endian raw import

unsimulatable, 2018-11-01:
Another field to select between little- and big-endian raws in the open data menu would be  helpful.

unsimulatable, 2018-11-03:
> @unsimulatable which camera or file format for pixelated STEM uses big
> endian raw data? Implementing this option makes sense if there's a
> specific use case.

Lieber Dieter Weber,

ich bin Benedikt Haas aus der AG von Christoph Koch; wir haben uns neulich gesehen. Die Anwendung ist eine Kamera an unserem modifizierten SEM (eine Lumenera Lu 170). Ich fände es praktisch, wenn ich die Datensätze nicht vorher konvertieren müsste. Aber falls die Implementierung aufwändig ist (was ich nicht vermutete) könnte ich nachvollziehen, dass es nicht gemacht würde, da unserer vermutlich doch ein eher seltener Fall ist.

Beste Grüße, 
Benedikt Haas

sk1p, 2018-11-03:
Implementation should be more or less straight-forward, reading big-endian files will have some performance impact though. Scheduling this for the 0.2 milestone, as we are very close to releasing 0.1. 

Let me know if you'd like to have a try at implementing it yourself!

uellue, 2018-11-03:
@unsimulatable @sk1p Sounds good! :+1:

uellue, 2018-11-04:
By the way, fast endian conversion could be a nice first issue to try some performance engineering. The operation is simple enough and should be very, very fast if it is implemented right. The approach would be to test if built-in methods are fast enough, and if not build a faster one with numba.

sk1p, 2018-11-12:
Hi @unsimulatable - we just realized that this is already possible given the current raw reader! You can use the standard numpy way to denote endianess, for example for big-endian 16bit unsigned ints, use `>u16`. See [the numpy documentation](https://docs.scipy.org/doc/numpy-1.15.1/reference/arrays.dtypes.html) for details on dtype specification. Note that there was a bug in an unrelated code path that prevented endian-annotated dtypes from working, which I just fixed in 1645904231f36671d47fc15d95ff1582996695dd, so in practice this only works with the latest development release.

So, the point now would be to add this to our documentation so that it doesn't get lost.

The performance overhead should not be so bad, but I didn't benchmark it in any comprehensive way.

Please, let me know if this works for you or if you need any more assistance!

unsimulatable, 2018-11-16:
Hi @sk1p, I tried it with the latest release version (the one I obtained by updating via pip) and using the correct expression for big-endian unsigned 16bit (which actually seems to be >u2) and I got an error (the message read just '>u2'). In comparison, if I just type something that is not a data type into the data type field, like xyz, the error message reads 'data type "xyz" not understood'.

sk1p, 2018-11-16:
Yes, `>u2` is correct, sorry. Installing via pip (`pip install libertem`) gives you the latest stable release, 0.1, which does not yet have this bug fixed. You need the current development version for that to work, which means either installing via git (as [described in our documentation](https://libertem.github.io/LiberTEM/install.html#installing-from-a-git-clone)), or using [the continuously built AppImage (Linux only)](https://github.com/LiberTEM/LiberTEM/releases/tag/continuous). If you'd like to keep up-to-date with current developments, keeping a local git clone is preferrable. Let me know if this works for you!

unsimulatable, 2018-11-19:
Yes, it works using the git clone. Thanks, problem solved!

PR #162 (2018-11-20): Included big endian example in RAW opening dialog
Include a numpy dtype example for big endian raw data in the RAW file opening dialog. Fixes #144

This commit includes fixes in setup.py to allow building the JS client on Windows. Credit goes to @sk1p !
commit 300dab19 (2018-11-20): Included big endian example in RAW opening dialog

Include a numpy dtype example for big endian raw data in the RAW file opening dialog. Fixes #144

This commit includes fixes in setup.py to allow building the JS client on Windows.
closed 2018-11-20 by sk1p