LiberTEM/LiberTEM #154: Dark frame substraction support

sk1p, 2018-11-13:
* calculation: sum frames job over a list of frames, or given as input file
 * efficiency: keep the dark frame on each worker node (how? have detailed look at dask distributed API)

sk1p, 2018-12-12:
The low-level `DataSet` API needs to communicate the need to start a setup callable/job/task. Usually, the input data to get the dark frame should be quite small, so we have to see if it makes sense to divide it into multiple `Task`s, or if one is enough, or we can decide based on number of frames to process.

The API can then, on `load`, ask the `DataSet` if any setup is needed, and run the setup task(s) with the current executor. The results can then be fed back to the `DataSet`, or later to a new `JobContext` API that scatters some shared static data into the cluster.

uellue, 2018-12-17:
It should be possible to deactivate automatic correction, for example for #155

uellue, 2018-12-17:
See also #201

sk1p, 2019-01-23:
FRMS6 support is now merged, with dark frame substraction support. As the dark frame data that is bundled with each FRMS6 dataset is just a few hundred frames, right now the dark frame is calculated on `initialize`, which takes the loading role described above (see also  https://github.com/LiberTEM/LiberTEM/commit/f38e7831d4be4700466c233db5b0fb322571e0b1).

uellue, 2020-05-04:
Adding to 0.6 milestone as part of #153