LiberTEM/LiberTEM #156: Documentation for the file formats

uellue, 2018-11-13:
Write detailed documentation on how to open the various supported formats. Should be done after improving usability of opening files #65 .

- [ ] MIB #129 
- [ ] RAW (different variants, including endian-ness #144 ,can easily be generated)
- [ ] HDF5 (can easily be generated)
- [ ] BLO 
- [ ] K2IS #63 #109 
- [ ] FRMS6  #155
- [ ] MRC #152 
- [ ] Norpix Seq #153

sk1p, 2019-06-25:
I've started working on this, see https://libertem.github.io/LiberTEM/formats.html in a few minutes. Let me know what you think about this style of docs. For some formats, in addition to the docstring, we may need a paragraph about differences to the GUI (for example, the GUI cannot specify the gain map for FRMS6 data sets yet)

sk1p, 2019-08-26:
As we have neither MRC nor Norpix Seq support yet, I'll close this for now. New file formats should be directly added to the docs.

closed 2019-08-26 by sk1p