LiberTEM/LiberTEM #181: Basic tests for Web API

uellue, 2018-11-28:
With the experience of #180 and #179 it would be desirable to have at least basic test coverage for the Web API. That would have caught these bugs. In particular testing both on Linux and Windows on a clean install is a great help.

sk1p, 2018-11-28:
Relevant projects:

 - https://github.com/eugeniy/pytest-tornado - sadly not maintained currently
 - https://github.com/pytest-dev/pytest-asyncio - this could help to start a temporary test server
 - https://www.tornadoweb.org/en/stable/testing.html - built in testing support of tornado using the unittest framework, should be runnable via pytest, but means writing the test cases differently

My current idea is to implement something using pytest-asyncio, which then is also less dependent on tornado, should we get the idea of using a different framework for the web API

sk1p, 2018-11-28:
There is an [active fork of pytest-tornado](https://github.com/vidartf/pytest-tornado) that is published at https://pypi.org/project/pytest-tornado5/ - this may be another option that could be more straight-forward than using pytest-asyncio. As an example project using pytest-tornado (not the fork though) I looked at jupyterhub:

 - fixture to create a mock tornado app: https://github.com/jupyterhub/jupyterhub/blob/9a3dbedc52b1894b08c4a7fa9099e20b3f9a3f69/jupyterhub/tests/conftest.py#L63
- example tests: https://github.com/jupyterhub/jupyterhub/blob/9a3dbedc52b1894b08c4a7fa9099e20b3f9a3f69/jupyterhub/tests/test_pages.py

uellue, 2018-11-28:
Looks simple enough! :-)

sk1p, 2019-01-02:
Maintenance discussion: https://github.com/eugeniy/pytest-tornado/issues/41

commit c300b6a4 (2019-03-14): basic async tests for web API, closes #181
closed 2019-03-14 by sk1p