LiberTEM/LiberTEM #182: Slow tests

sk1p, 2018-11-28:
There are two tests that take much longer than all others (output of `tox -e py36 -- --durations=5`):

```
========================================================================= slowest 5 test durations ==========================================================================
4.56s call     tests/test_analysis_masks.py::test_weird_partition_shapes_1
1.11s call     tests/test_analysis_masks.py::test_subframe_tiles
0.14s call     tests/io/test_k2is_uint12.py::test_encode_decode_uint12_ref
0.14s call     tests/io/test_k2is_uint12.py::test_encode_decode_uint12
0.12s call     tests/test_analysis_masks.py::test_mask_uint
========================================================================= 58 passed in 8.27 seconds =========================================================================
```

`test_weird_partition_shapes_1` runs three mask analyses, each over 64 quite small partitions. In 

`test_subframe_tiles` it is just many small tiles.

This needs to be profiled and then we can decide if this is a performance bug in mask application or if we should reduce the number of partitions for improving testing speed.

sk1p, 2018-12-06:
I did some profiling on this, as the slow tests annoyed me quite a bit.

I used `line_profiler` for this, with the following methodology:

1) Install `line_profiler` via pip. 
2) Insert the `@profile` decorator on the function you are interested in
3) Run the test case you are interested in with the profiler, for example: `$ kernprof -l -v $(which pytest) tests/test_analysis_masks.py::test_weird_partition_shapes_1`
4) Move the decorator down to hot function you are interested in

The profile decorator is injected into the builtins, so it is gobally available without importing anything.

sk1p, 2018-12-06:
Some hotspots for `test_weird_partition_shapes_1`:

```
Total time: 12.5349 s
File: /tmp/libertem/src/libertem/job/masks.py
Function: _get_masks_for_slice at line 16

Line #      Hits         Time  Per Hit   % Time  Line Contents
==============================================================
    16                                               @functools.lru_cache(maxsize=None)
    17                                               @profile
    18                                               def _get_masks_for_slice(slice_):
    19                                                   """
    20                                                   """
    21                                                   sliced_masks = [
    22                                                       # .reshape((-1, 1)) -> like flatten, but compatible with sparse
    23                                                       # matrices and no copies
    24                                                       # should save us one copy as we np.hstack() immediately afterwards
    25                                                       # https://stackoverflow.com/a/28930580/540644
    26     49152      40554.0      0.8      0.3              slice_.get(mask, sig_only=True).reshape((-1, 1))
    27     49152    7805436.0    158.8     62.3              for mask in computed_masks
    28                                                   ]
    29                                                   # MaskContainer assures that all or none of the masks are sparse
    30     49152      79004.0      1.6      0.6          if sp.issparse(sliced_masks[0]):
    31     16384    4373460.0    266.9     34.9              return sp.hstack(sliced_masks)
    32                                                   else:
    33     32768     236409.0      7.2      1.9              return np.hstack(sliced_masks)

Total time: 19.975 s
File: /tmp/libertem/src/libertem/job/masks.py
Function: __call__ at line 217

Line #      Hits         Time  Per Hit   % Time  Line Contents
==============================================================
   217                                               @profile
   218                                               def __call__(self):
   219       192        383.0      2.0      0.0          num_masks = len(self.masks)
   220       192       1641.0      8.5      0.0          part = np.zeros((num_masks,) + self.partition.shape[:2], dtype="float32")
   221     49344    1612143.0     32.7      8.1          for data_tile in self.partition.get_tiles():
   222     49152     518675.0     10.6      2.6              data = data_tile.flat_data
   223     49152      51475.0      1.0      0.3              if data.dtype.kind == 'u':
   224     49152     169027.0      3.4      0.8                  data = data.astype("float32")
   225     49152   13119840.0    266.9     65.7              masks = self.masks[data_tile]
   226     49152      48173.0      1.0      0.2              if self.masks.use_sparse:
   227                                                           # The sparse matrix has to be the left-hand side, for that
   228                                                           # reason we transpose before and after multiplication.
   229     16384    1542875.0     94.2      7.7                  result = masks.T.dot(data.T).T
   230     32768      20071.0      0.6      0.1              elif self.use_torch and torch is not None:
   231     32768      19600.0      0.6      0.1                  result = torch.mm(
   232     32768     116810.0      3.6      0.6                      torch.from_numpy(data),
   233     32768     332687.0     10.2      1.7                      torch.from_numpy(masks),
   234                                                           ).numpy()
   235                                                       else:
   236                                                           result = data.dot(masks)
   237     49152      40692.0      0.8      0.2              self.copy_to_part_result(
   238     49152     455439.0      9.3      2.3                  tile_slice=data_tile.tile_slice.shift(self.partition.slice),
   239     49152      29025.0      0.6      0.1                  tile_result=result,
   240     49152    1894036.0     38.5      9.5                  dest=part
   241                                                       )
   242                                                   return [
   243       192        102.0      0.5      0.0              ResultTile(
   244       192        105.0      0.5      0.0                  data=part,
   245       192        146.0      0.8      0.0                  partition_slice=self.partition.slice,
   246       192       2024.0     10.5      0.0                  dest_slice=self._get_dest_slice_4d(self.partition.slice),
   247                                                       )
   248                                                   ]

Total time: 20.3659 s
File: /tmp/libertem/tests/test_analysis_masks.py
Function: _run_mask_test_program at line 12
[... 8 lines cut ...]
    17         1          1.0      1.0      0.0      analysis_sparse = lt_ctx.create_mask_analysis(
    18         1          3.0      3.0      0.0          dataset=dataset, factories=[lambda: to_sparse(mask)], use_sparse=True
    19                                               )
    20         1          1.0      1.0      0.0      analysis_dense = lt_ctx.create_mask_analysis(
[... cut ...]
    33                                                   expected
    34                                               )
    35         1         47.0     47.0      0.0      assert np.allclose(
    36                                                   results_dense.mask_0.raw_data,
    37                                                   expected
    38                                               )
```

I'll post this without comment for now, also note that this is with some changes on top of the master branch, so take it with a grain of salt...

sk1p, 2018-12-06:
Same profile spots for `test_subframe_tiles`:

```
Total time: 3.23698 s
File: /tmp/libertem/src/libertem/job/masks.py
Function: _get_masks_for_slice at line 16

Line #      Hits         Time  Per Hit   % Time  Line Contents
==============================================================
    16                                               @functools.lru_cache(maxsize=None)
    17                                               @profile
    18                                               def _get_masks_for_slice(slice_):
    19                                                   """
    20                                                   """
    21                                                   sliced_masks = [
    22                                                       # .reshape((-1, 1)) -> like flatten, but compatible with sparse
    23                                                       # matrices and no copies
    24                                                       # should save us one copy as we np.hstack() immediately afterwards
    25                                                       # https://stackoverflow.com/a/28930580/540644
    26     12288       9805.0      0.8      0.3              slice_.get(mask, sig_only=True).reshape((-1, 1))
    27     12288    2001276.0    162.9     61.8              for mask in computed_masks
    28                                                   ]
    29                                                   # MaskContainer assures that all or none of the masks are sparse
    30     12288      19057.0      1.6      0.6          if sp.issparse(sliced_masks[0]):
    31      4096    1144782.0    279.5     35.4              return sp.hstack(sliced_masks)
    32                                                   else:
    33      8192      62057.0      7.6      1.9              return np.hstack(sliced_masks)

Total time: 5.13347 s
File: /tmp/libertem/src/libertem/job/masks.py
Function: __call__ at line 217

Line #      Hits         Time  Per Hit   % Time  Line Contents
==============================================================
   217                                               @profile
   218                                               def __call__(self):
   219         3         26.0      8.7      0.0          num_masks = len(self.masks)
   220         3         23.0      7.7      0.0          part = np.zeros((num_masks,) + self.partition.shape[:2], dtype="float32")
   221     12291     411754.0     33.5      8.0          for data_tile in self.partition.get_tiles():
   222     12288     132102.0     10.8      2.6              data = data_tile.flat_data
   223     12288      12553.0      1.0      0.2              if data.dtype.kind == 'u':
   224     12288      43426.0      3.5      0.8                  data = data.astype("float32")
   225     12288    3380444.0    275.1     65.9              masks = self.masks[data_tile]
   226     12288      11605.0      0.9      0.2              if self.masks.use_sparse:
   227                                                           # The sparse matrix has to be the left-hand side, for that
   228                                                           # reason we transpose before and after multiplication.
   229      4096     405268.0     98.9      7.9                  result = masks.T.dot(data.T).T
   230      8192       4962.0      0.6      0.1              elif self.use_torch and torch is not None:
   231      8192       4687.0      0.6      0.1                  result = torch.mm(
   232      8192      27743.0      3.4      0.5                      torch.from_numpy(data),
   233      8192      84637.0     10.3      1.6                      torch.from_numpy(masks),
   234                                                           ).numpy()
   235                                                       else:
   236                                                           result = data.dot(masks)
   237     12288       9988.0      0.8      0.2              self.copy_to_part_result(
   238     12288     114697.0      9.3      2.2                  tile_slice=data_tile.tile_slice.shift(self.partition.slice),
   239     12288       7203.0      0.6      0.1                  tile_result=result,
   240     12288     482308.0     39.3      9.4                  dest=part
   241                                                       )
   242                                                   return [
   243         3          2.0      0.7      0.0              ResultTile(
   244         3          3.0      1.0      0.0                  data=part,
   245         3          2.0      0.7      0.0                  partition_slice=self.partition.slice,
   246         3         34.0     11.3      0.0                  dest_slice=self._get_dest_slice_4d(self.partition.slice),
   247                                                       )
   248                                                   ]

Total time: 5.22845 s
File: /tmp/libertem/tests/test_analysis_masks.py
Function: _run_mask_test_program at line 12
[... 6 lines cut ...]
    15         1          9.0      9.0      0.0          dataset=dataset, factories=[lambda: mask]
    16                                               )
    17         1          1.0      1.0      0.0      analysis_sparse = lt_ctx.create_mask_analysis(
    18         1          3.0      3.0      0.0          dataset=dataset, factories=[lambda: to_sparse(mask)], use_sparse=True
[... cut ...]
    31         1         56.0     56.0      0.0      assert np.allclose(
    32                                                   results_sparse.mask_0.raw_data,
    33                                                   expected
    34                                               )
    35         1         50.0     50.0      0.0      assert np.allclose(
    36                                                   results_dense.mask_0.raw_data,
    37                                                   expected
    38                                               )
```

sk1p, 2018-12-06:
To make the tests run in an acceptable time again, I'm going to do the following:

1) make both a "slow" and a "fast" variant for the tests
2) "fix" the fast variant by dividing into less partitions / tiles
3) keep the slow variant around for future profiling sessions

For future profiling, the slow tests can be run with `pytest -m slow` (by default, `-m "not slow"` will be added to the command line via `setup.cfg`)

uellue, 2018-12-11:
@sk1p the tests with sparse matrices are executed not only for matrices that are actually sparse, but for densely populated matrices as well. That is not efficient at all, as confirmed by the profiling.

I just included these tests to confirm correct functionality in all border cases. If the sparse test slows down tests where we can be SURE that the sparse version will work the same, we can restrict using sparse to tests where it performs well. Let me know if I should change something!

sk1p, 2018-12-11:
Using less partitions or tiles works well for speeding up the tests, should be fine once #195 is merged.

Also keeping these around allows us to benchmark these worst-cases, and may be helpful if we decide to use https://github.com/pydata/sparse in the future.

PR #195 (2018-12-12): Dimensionality refactoring
First part of making LiberTEM more flexible regarding dimensionality of data. Introduces a `Shape` class that knows how many dimensions the signal has, a bit like hyperspy (simpler, keeping numpy ordering of dimensions etc.).

Tests should pass, but support in the I/O layer is not complete yet.

Closes #182, #157, #99
closed 2018-12-12 by uellue