LiberTEM/LiberTEM #186: Scheduling of background and high-performance tasks

uellue, 2018-12-05:
LiberTEM can open many data sets in parallel. Some of the data might be in the file system cache and/or on fast local storage. Other data might be in other locations with far less bandwidth. Even a 10 GbE NAS or a regular SSD can be regarded as very slow from a LiberTEM perspective.

Right now, tasks are processed on a "first come, first serve" basis by one worker process per CPU core. This can be inefficient in such a heterogeneous scenario because the IO-bound slow process is waiting for a long list of high-performance tasks or vice-versa, although they can be efficiently processed in parallel by the operating system. 

Proposed solution (discussion with @sk1p): Start a swarm of workers with one worker per core individually for each data set. Concurrency between data sets is then handled by the operating system. Disadvantage: Two high-performance data sets would interfere with each other.

Perhaps distinguish by data set source, i.e. a swarm of workers per file system, not data set?

uellue, 2018-12-13:
Maybe include checks if a block is in the file system cache? http://man7.org/linux/man-pages/man1/fincore.1.html For example, have dedicated workers for cached files that will not load IO, but heavily load memory and CPU.

For Windows I couldn't find an equivalent to fincore yet.

sk1p, 2018-12-13:
We can use the `dask.distributed` priority system, instead of having more workers. This already works well for file browsing etc. 

The question then is: can we extend that to more "fuzzy" priorities? Like you say, we can determine if a dataset is in fs cache, or how much of it is in fs cache, but can we somehow guarantee fairness for datasets? Like: one dataset is in cache, it has higher priority, but another one is doing slower I/O. The job for the slow dataset should have at least one worker running on each node.

Maybe we should defer this one until after we have a real cluster running, anything else is just a thought experiment or guessing...

uellue, 2018-12-13:
@sk1p 👍 for testing! Actually, the point here is that the IO-heavy and CPU-heavy tasks can run efficiently in _parallel_., using hyperthreading and the scheduling of the operating system. What about one worker per CPU for cached data, plus a single one for each slow data source? One or two workers should even saturate a 10GbE network connection or a SATA SSD.

sk1p, 2018-12-14:
Another scheduling and I/O-speed related problem is choosing the right partition size. Right now, if the data is loaded from a slow source, the partitions are too large for interactive feedback. A possible solution would be to have adaptive partition sizes, and to determine the speed by starting with a very small partition, and scaling up until we reach some predetermined latency (~300ms?). Kind of like TCP window size...

uellue, 2019-09-24:
Another point is allowing file browsing while a computation is running. Currently, the file browser lags/freezes while a computation is running because it needs to run code on the server via the same workers that do the computation.

In the future, tasks like file browsing and checking datasets should get dedicated workers to ensure fast responses.

uellue, 2023-08-15:
As a comment, we have service workers in a Dask cluster now that improve the file browsing responsiveness. We support threaded workers, including a mixture of threads and processes. Also, since we don't schedule all tasks at once but fill up the queue gradually so that tasks are scheduled on available workers, it would be possible to restrict the number of in-flight tasks per UDF run, i.e. have extra workers available without automatically oversubscribing. However, I am still not aware of a way to schedule I/O and computation independent of each other.

uellue, 2024-05-16:
This hasn't been a big issue since we process data from only one source at a time in real-world use cases. Closing until this comes up as a real-world issue.

closed 2024-05-16 by uellue