LiberTEM/LiberTEM #188: Array Alignment

robbmcleod, 2018-12-06:
@sk1p asked on the Hyperspy IO thread about data alignment importance. Since it's an involved question, I'll just raise an issue here instead.

> For alignment, maybe the following is of interest: https://lemire.me/blog/2012/05/31/data-alignment-for-speed-myth-or-reality/ - have you made different experiences?

I would say that blog is not showing us the assembly that is generated. Presenting a benchmark and not declaring how you compiled it is a bit insufficient. We would have to see the results from the compiler's effort at auto-vectorization. In 2012, there's no way to be sure from the blog post if the author was using a compiler that could auto-vectorize that templated code. I mean, `i7` is awfully generic way to describe a processor. Also in 2012 there weren't that many integer-based SIMD instructions. It would be more useful to talk about float data. 

The compilers all have fairly specific use cases where they can vectorize loops or not. Generally from best to worst,

`icc`
`clang`
`gcc`
`cl`

Pointer math, for example, generally isn't vectorizable. Typically with `gcc` one wants to compile with `-fopt-info-vec` and possibly `-fopt-info-vec-missed`. With MSVC you can call with `/Qvec-report:2`

There have been some efforts to provide aligned memory in Python. First, Raymond Hettinger:

https://bugs.python.org/issue18835

Second, Antoine Pitou (who is also a Python core dev as well as being one of the core Numba devs) made a couple of PRs for NumPy:

https://github.com/numpy/numpy/issues/5312

There's whitepapers from Intel linked in the NumPy issue that suggest 256-bit alignment for AVX2 and 512-bit for AVX512 instructions. They also linked `numba` benchmarks that showed a 40 % speed difference between aligned versus not-aligned. 

Both of those issues have not resulted in changes. My impression this is because the Windows implementation faults if you call `free` on memory reserved via `_aligned_malloc`. I have thought about raising this on the `numpy-dev` mailing list and see if there's any more interest.

sk1p, 2018-12-11:
Thanks for the comment and links! 40% improvement sounds like something that is worth working towards, also considering portability to CPUs that may or may not be slow in case of unaligned access.

For mmapped I/O, the data should already be properly aligned, at least on Linux, as the address is guaranteed to be on a page boundary. Of course this works only as long as the header sizes of the given format are multiples of 64 bytes. This is the case for MIB files, for example (6*64 bytes IIRC), but sadly not for K2IS datasets, where the data block header is 40 bytes long (that saves about 6kb per frame, yay!).

It should be possible to align numpy arrays "by hand", i.e. over-allocate by 64 bytes and create a `view` that starts at an aligned address. Would need to benchmark this and maybe have a look at numpy internals if this even has a chance to work.

I think, in the context of LiberTEM, it makes sense to defer this work until we have a proper benchmark suite (#198 ). For designing a file format this is definitely interesting and relevant!

robbmcleod, 2018-12-11:
I know how to memory-align a 1D array in NumPy:

https://github.com/robbmcleod/pyfastnoisesimd/blob/master/pyfastnoisesimd/helpers.py

It's fiendishly difficult to align a 2D array unless each x-axis shape is a multiple of the SIMD length. If you are doing element-wise operations its no problem if a SIMD operation spans two rows, but if doing row-wise operations it would write past the row.  You can't get a view from NumPy with hidden zeros at the end of each row as far as I can tell; as I recall it returns a copy. So a 256 x 256 detector would be fine but something with a strange pixel shape like the K2 is not trivial.

I also don't know how to do it inside a C-extension, only in Python. The NumPy multiarray is quite a diffuse structure spread out over a lot of files.