LiberTEM/LiberTEM #197: Compare sparse matrix implementations

uellue, 2018-12-11:
Currently, LiberTEM uses scipy.sparse for sparse matrix multiplication. 

However, the alternative implementations from [torch](https://pytorch.org/docs/stable/sparse.html) and [pydata](http://sparse.pydata.org/en/latest/) are available as well.

We should compare the implementations with regards to performance, stability and ease of use, and decide if we should stick with scipy.sparse, pick something else or even support several implementations like we do with dense matrices already.

See also #182

uellue, 2019-05-13:
https://sparse.pydata.org/ is coming out ahead by a large margin in most cases:

```python
import numpy as np
import scipy.sparse as sp
import sparse
import torch

# 1 MiB tile size
size = 1024*1024*1
​
(fy, fx) = (512, 512)
​
l = size // (fy*fx)
​
print(l)
# 4
​
data = np.random.random((l, fy*fx))
​
n_masks = 100
n_entries = 25
​
masks = np.zeros((n_masks, (fy*fx)))
​
mask_indices = []
frame_indices = []
values = []
​
for i in range(100):
    for y in np.random.choice(range(fy), size=n_entries, replace=False):
        for x in np.random.choice(range(fx), size=n_entries, replace=False):
            index = y*fx + x
            mask_indices.append(i)
            frame_indices.append(index)
            masks[i, index] = 1
            values.append(1)

np.dot(data, masks.T)

%timeit np.dot(data, masks.T)
# 51.6 ms ± 963 µs per loop (mean ± std. dev. of 7 runs, 10 loops each)

t_data = torch.from_numpy(data)
t_masks = torch.from_numpy(masks.T)
torch.mm(t_data, t_masks)

%timeit torch.mm(t_data, t_masks)
# 39.2 ms ± 1.95 ms per loop (mean ± std. dev. of 7 runs, 10 loops each)

sp_masks = sp.csr_matrix(masks)
sp_masks.dot(data.T)

%timeit sp_masks.dot(data.T)
# 13.4 ms ± 419 µs per loop (mean ± std. dev. of 7 runs, 100 loops each)

i = torch.LongTensor([mask_indices, frame_indices])
v = torch.DoubleTensor(values)
t_sp_mask = torch.sparse.DoubleTensor(i, v, torch.Size([n_masks, fy*fx]))
t_sp_data = torch.from_numpy(data.T)
torch.sparse.mm(t_sp_mask, t_sp_data)

%timeit torch.sparse.mm(t_sp_mask, t_sp_data)
#334 29.1 ms ± 1.54 ms per loop (mean ± std. dev. of 7 runs, 10 loops each)

pydata_sp_mask = sparse.COO([mask_indices, frame_indices], values, shape=(n_masks, fy*fx))
sparse.dot(pydata_sp_mask, data.T)

%timeit sparse.dot(pydata_sp_mask, data.T)
#4.09 ms ± 76.3 µs per loop (mean ± std. dev. of 7 runs, 100 loops each)
```

Further tests have revealed that sparse matrices can be competitive at up to 20 % occupation, roughly, which makes them an attractive choice for quite a wide range of scenarios. https://sparse.pydata.org/ was leading or at least neck and neck when the parameters were varied.

https://sparse.pydata.org/ is a bening dependency for us because it doesn't introduce new sub-dependencies.

In conclusion, we should use https://sparse.pydata.org/

I am renaming the issue to swap out the current implementation.

uellue, 2019-05-14:
Here's a development branch that uses sparse.pydata.org: https://github.com/uellue/LiberTEM/tree/sparse

However, testing revealed that the performance advantage is killed by compilation time, about 1.5 s on my machine, since we create a new sparse array for each run and trigger a recompilation. Is there a way around that?

It seems that the recompilation time is independent of the matrix size. That means scipy.sparse would come out ahead for shallow mask stacks and small data sets, while sparse.pydata.org comes out ahead for large data sets and deep mask stacks.

uellue, 2019-05-14:
A bit more testing revealed that likely the sparse.dot() method is recompiled in each run, not because we send a new matrix, but because we send the dot() method as part of the future. That should be possible to avoid, I guess?

uellue, 2019-05-14:
Update: executing sparse.dot() on the server and/or the workers through libertem.preload before sending futures didn't work.

uellue, 2019-05-14:
PR #336 now contains a notebook https://github.com/LiberTEM/LiberTEM/blob/1f2909dd34192eb827afe72e2fbd7f2173a8166d/prototypes/sparse-comparison/performance-recompilation.ipynb that compares the effects.

However, the results are quite different when doing the same calculation in the GUI. On the same data set, sparse is slower.

sk1p, 2019-05-15:
> the results are quite different when doing the same calculation in the GUI

Hmm, the measurement done when using the GUI is not very precise; it is done on the client side (to include network overheads etc.). I trust the measurements done via the Python API much more.

uellue, 2019-05-15:
Ok, so that should mean we are in business? By the way, on moellenstedt reading from RAW sparse brings an 1.5x advantage for small disks or rings. I guess deep mask stacks will be strongest for sparse.

sk1p, 2019-05-15:
> that should mean we are in business

I'd say so. If it works well with the Python API, I consider the performance difference to the GUI a problem with the web API or async executor or async adapter.

commit b1a5ce3e (2019-05-16): Use sparse.pydata.org instead of scipy.sparse

Backwards compatibility with scipy.sparse is kept.

See #197
Closes #197
closed 2019-05-16 by sk1p