LiberTEM/LiberTEM #198: Well-organized benchmarking suite

uellue, 2018-12-11:
Right now, we do have automated tests, but we don't have automated benchmarks. 

We should have a benchmarking suite that can be executed to gauge the impact of a change. The unit tests are not really suitable because they often use LiberTEM in a non-standard way: inline executor instead of dask, very small in-memory data sets, test of "weird" partitioning schemes, using sparse matrices where one really shouldn't #182 .

Instead, we should have a reliable test program with the following properties:

* Test on large real-world data sets with typical analyses
* Executed under reproducible conditions: Dedicated machine without background tasks
* Executed on a small cluster as well as a typical single-node workstation.
* Collect profiling information
* Regular automated collection and reporting of benchmark results to catch changes with a negative impact on performance.

sk1p, 2018-12-11:
See also: https://github.com/airspeed-velocity/asv

uellue, 2019-01-08:
The benchmarking suite should separate processing and overheads. Overheads can be a major factor #232

uellue, 2019-05-28:
We should routinely compare math libraries:

* Torch vs openblas (vs mkl?)
* pyfftw vs numpy.fft vs scipy.fft (vs mkl?)

As a side note, it seems that openblas might have caught up to torch recently -- further tests desirable.

sk1p, 2019-06-03:
Btw. for FFT comparison, see also https://www.kcod.de/gearshifft/

sk1p, 2020-01-30:
Idea: use an Azure on-premise Agent to run the benchmarks as part of the CI pipeline, on a dedicated server. Start with tests that can run in a few seconds, maybe later introduce daily benchmark run for longer running tests (for example ones that create some TB of data on demand, also to test our writing facilities - things you don't want to run on every commit)

uellue, 2020-06-17:
Asvlooks like the best bet so far and should cover all points when used appropriately.

A few more detailed thoughts on use cases and intermediate results after playing with asv:

* Routinely check during development if a change introduced regressions to catch things like #814 before merging
  - That means asv has to run code from current fork, current branch.
  - Using the "same repo" layout collides with the existing structures of LiberTEM.
  - Using the "separate repo" layout in a separate repo LiberTEM/benchmarks required setting a Git URL or path to the actual repo to benchmark, meaning one would have to merge into main repo master first or adapt the path to the way how the repos are checked out, which is fragile.
  - Using the "separate repo" layout in a subfolder of the main repo, and setting the repo URL/path to a relative path pointing to the repository root ('..') worked.
* Many benchmarks and required fixtures are equivalent to existing unit tests
  - We should make sure to re-use as much code as possible for both to avoid duplication
  - In particular the files for dataset tests are interesting!
  - The fixture mechanism and code loading of pytest is rather specific and I couldn't find documentation on how to use fixtures and run individual test functions without running the entire pytest suite. Calling a fixture directly fails complaining loudly: https://docs.pytest.org/en/latest/deprecations.html#calling-fixtures-directly
  - In an ideal world, the functionality of asv to record and track timings as a function of commit, machine, config etc would be folded into pytest...
  - As a way out, perhaps one can define common utility functions in such a way that they can work within the asv benchmarks and wrap them into fixtures for pytest? Import or wrap some common test functions from that code base rather than defining them in the tests folder?
    + What would be a good place for common infrastructure? Within module of libertem package (easy to import), or repo subfolder (keep infra out of release, but harder to import)?
    +  Some functions may actually fit well in a separate repo (LiberTEM-infra?) since LiberTEM sub-packages like blobfinder duplicate them as well.
* We'll have to set up gitignore rules to mask out a lot of the intermediate data of asv (envs, HTML, ...)
* We'll have to figure out a process when to commit which parts of the asv directory structure to save benchmarks, and how to discard unwanted ones, making sure they are not committed.
* We'll have to think how to integrate the benchmarks in the gh-pages structure. Fortunately it seems that the html, css, js output of asv is relocatable -- to be verified!

uellue, 2020-06-17:
Specific item: Test if the resource check and other parts of the the DaskJobExecutor scale to at least 1000 workers, better 10000.

uellue, 2020-06-17:
Issue: Separate first run from repeat runs (Numba compile time etc)!

uellue, 2020-06-18:
This one actually extends pytest and checks many of the items as well: https://pytest-benchmark.readthedocs.io/en/latest/index.html

sk1p, 2020-06-18:
> This one actually extends pytest and checks many of the requirements as well: https://pytest-benchmark.readthedocs.io/en/latest/index.html

Nice! As I understand, this is missing the reporting functionality of airspeed velocity - if you could write up how that should work I could write a prototype for that.

sk1p, 2020-06-18:
Or is plotting using `--benchmark-histogram` enough for our purposes?

uellue, 2020-06-18:
Yes, it only has basic reporting. Since the data is available as JSON, it would not be too difficult to write a custom reporting function that discovers the files, extracts the relevant data and plots it, for example with matplotlib. We could call that as part of the build process like we do with `build-authors-contributors`.

uellue, 2020-06-18:
...cont: Advantage would be that we have full control what is plotted how and how tests are grouped, including extracting the first run vs. repeats, overlay multiple plots for comparison, custom units such as pix/s, etc. Disadvantage is of course more work to make it look nice. IMO a trade-off between cost of developing and maintaining benchmarks (lower with pytest) vs. cost of reporting.

sk1p, 2020-11-27:
As #565 is merged, we could now create a pipeline to run this regularly. As load on our on-prem agent varies, we would have to see what variances are introduced, and how well the iterations/rounds etc stabilizes the numbers.