LiberTEM/LiberTEM #199: API for remote operation and in different languages

uellue, 2018-12-11:
Discussions with @sk1p: Right now, the Python API has an implicit assumption that it is executed on a system equivalent to a worker node, i.e. it "sees" the same file system and has the same software versions. Furthermore, it directly uses the dask client.

This works perfectly fine for a local cluster and for using a cluster with a remote Jupyter notebook or other script that runs on the cluster's control node.

However, for integration in local software that uses a remote cluster, we have to remove this assumption.

1. It "sees" a different file system. That means tasks like checking if a data set can be created with the specified path have to be executed on the cluster, not on the local API client.
1. It can run a different version of LiberTEM and Python. We can't require users to replicate exactly the server environment. That means sending over functions with cloudpickle will not work well. That means we can't use a dask client in the local API to "talk" to the remote scheduler.
1. We should enable APIs in other languages than Python.
1. We need authentication that is both safe and convenient to use in a local application.

In essence, the API should probably use a form of RPC, REST, ... This is already done for the web API for standard analyses. Implementing arbitrary user functions to be executed on the cluster get more tricky than with the current setup, however.

Options would include:

* Sending over Python code snippets that are compiled on the remote system and then made available through the API --> needs debugging options and good error reporting
* Sending over code in other languages that is then compiled to machine code on the remote system and made available through, for example, ctypes similar to the way numba works. Perhaps support LLVM IL?

uellue, 2019-10-15:
Discussion with @sk1p: UDF interface implemented with sth like [WebAssembly](https://webassembly.org/) in order to support other languages and/or proprietary code for UDFs?

sk1p, 2020-01-09:
So, effectively we would have three different APIs:

 * cloudpickle-based interactive API (for example for Jupyter Notebooks, impl. via dask distributed for now). Good for exploration of new methods etc.
 * HTTP/websocket based API, which should be extensible by aplication modules. GUI applications could offload their computation/data reduction to LiberTEM by implementing specific functions as UDFs, which are shipped to the cluster nodes as part of the container image. May need some composability improvements, to run multiple UDFs on the same pass over the data.
 * As an extension to the HTTP API, a WebAssembly-based interface. This gives our users a way to write their UDFs in their favorite compiled language (C/C++/Rust should all work well), and also allows to re-use processing modules written in these languages, as long as they fit the computational model of LiberTEM.

The WebAssembly integration could be a nice candidate for GSoC (#543)

uellue, 2020-01-09:
RPyC https://rpyc.readthedocs.io/en/latest/ allows running commands on a different Python environment than the host.

uellue, 2020-01-09:
The Advanced Executor issue #473 could be related to this since we are working on overcoming the limits of sending over pickled Python functions. Pickling functions and their results is at the core of Dask futures and our current Executor.

uellue, 2023-10-25:
Closing in favor of a potential future integration effort that is more specific. See also https://www.sciencedirect.com/science/article/pii/S246806722300038X for an integration that might be related to such an API.

closed 2023-10-25 by uellue