LiberTEM/LiberTEM #200: Improve error reporting

uellue, 2018-12-11:
Discussion with @sk1p

* Show exception traceback in GUI
* General discussion point: Handle and report exceptions at the workers. They tend to get messy. Perhaps make very clear where an exception occurred, as a first step?

sk1p, 2018-12-13:
One way to improve debugging capabilities from the GUI would be to have a log functionality, that is driven by a `logging` handler implementation that sends the log over the websocket connection. This would include exceptions, and also log messages from other Python libraries.

For our own logging, we could later use https://github.com/hynek/structlog to make the logs more machine-readable and filterable in the client.

Handling exceptions on workers may actually not be the best idea - at least we need to re-throw them afterwards to properly fail the `Future`. The point would be to properly handle them at points where we retrieve the results from the cluster.

uellue, 2018-12-13:
@sk1p sounds good! We'll really need that as soon as we allow custom code.

With "handling" I meant dealing with them in a way that a user knows what happened. Right now it can be difficult to spot where and how the error happened if you don't understand the internal workings of dask.distributed and LiberTEM. It can be a collection of very long and confusing tracebacks when a load of futures trip up.  The minimum would be a clear information that it was a worker that encountered trouble. It would also help to not just dump all the tracebacks together, but to collapse them by default and generate a list where you can see "oh, ALL the workers threw an exception" and then pick out one to look at more closely.

uellue, 2025-03-26:
Not really relevant since one hardly sees errors in the GUI anymore. In Notebooks the traceback seems to work well enough.

closed 2025-03-26 by uellue