LiberTEM/LiberTEM #205: Do all filesystem operations via the executor

sk1p, 2018-12-12:
We want to do all filesystem operations via the executor. This actually has two reasons:

1) They may take some time, for example when listing a large directory, or listing a NFS share. Running them via the executor means we get to do this in the background.
2) The worker nodes may not have the same filesystem as the API server or `dask.distributed` `Client`

In the first iteration, this means we make sure `Context.load(...)` pushes the dataset load operation into the cluster, including the `check_valid()` call. The web API basically needs the same changes.

The next step would be to update the web API to also do directory listing / existence checks etc. via functions pushed into the cluster. 

This is highly related to #154 which also may need to do more work on dataset initialization. Also related is #186 as listing a directory and opening files should take priority over another job that may take a while to finish.

- [ ] add ability to run simpler functions via the executor
- [ ] update `Context` and `DataSetDetailHandler` to  do `load` and `check_valid` via executor
- [ ] test connecting via `Context` to a remote dask scheduler
- [ ] update `DataSetDetectHandler` 
- [ ] update `LocalFSBrowseHandler`
- [ ] test starting the web client with a connection to a remote dask scheduler

sk1p, 2018-12-13:
As of ad09ac8798ca72b13bf5ab67c603e5d38eeacd0c it is possible to spawn a dask cluster on a remote host, connect to it (for example by starting a LiberTEM server locally, and then connecting via GUI) and work with the files on the remote system.

Currently, this is a bit brittle, as certain methods on the dataset need to make sure they don't access the filesystem (they are used on the client to calculate partitioning etc.), and that the initialization function is called. If I get an idea how to structure this better, maybe it can be made more robust.

closed 2018-12-13 by sk1p

uellue, 2018-12-13:
Tested with the async Digital Micrograph prototype with processing on moellenstedt. It ROCKS! 👍 👍 👍