LiberTEM/LiberTEM #206: Support for "directory full of binary files" kind of DataSet

sk1p, 2018-12-12:
Some software, for example in STEM simulation, uses a directory full of small binary files as the result. We need to test if we can support this in a high-performance way, and if we can, add support in LiberTEM.

Generally, we may want to use straight `read` calls, as `mmap` should have more overhead, and we need to copy anyways if we want to stack the images for processing.

It should have similar parameters to the all-in-one-file raw format.

- [ ] implement prototype
- [ ] do basic benchmarking with synthetic data via the Python API
- [ ] if it performs, clean up implementation and see if we want to add it to the UI

sk1p, 2018-12-19:
Test data set: split up a 4G dataset into individual frames (64k per frame), each in a single file. This is basically the worst case.

Small test script:

```python
import os
os.environ['OMP_NUM_THREADS'] = '1'
os.environ['MKL_NUM_THREADS'] = '1'
os.environ['OPENBLAS_NUM_THREADS'] = '1'

import numpy as np
from libertem import api
from libertem.executor.inline import InlineJobExecutor
from libertem.io.dataset.raw_parts import RawFilesDataSet

@profile
def main():
    ctx = api.Context(executor=InlineJobExecutor())

    ds = RawFilesDataSet(
        path="/home/clausen/Data/many_small_files/frame00016293.bin",
        scan_size=(256, 256),
        detector_size=(128, 128),
        tileshape=(1, 8, 128, 128),
        dtype="float32"
    )
    ds.initialize()

    job = ctx.create_mask_analysis(dataset=ds, factories=[lambda: np.ones(ds.shape.sig)])

    result = ctx.run(job)

if __name__ == "__main__":
    main()
```

Profiling results (raw results below):

 + `glob.glob` takes quite a while, but that is initialization cost and easily amortized
 + most of the job time is indeed spent in I/O (`get_tiles(...)`)
 + in `get_tiles`, ~20% are spent opening files, ~20% copying to the tile buffer, and ~45% reading using `np.fromfile`. The copy may be avoidable, by somehow reading directly into the tile buffer, but opening and reading cannot be avoided while keeping to the tiling scheme

Notes: 
 + we get more overhead by needing to read from more than one file for stacking (copies...)
 + it may actually be more efficient in this case to just use smaller tiles to get rid of copies, did not try yet
 + reading many small files means we get more filesystem overhead, and no longer read in large contiguous blocks

Line profiling results for reading many small files:

```
/home/clausen/Data/many_small_files/frame*.bin
Wrote profile results to raw_many_files.py.lprof
Timer unit: 1e-06 s

Total time: 3.27343 s
File: /home/clausen/source/libertem/src/libertem/io/dataset/raw_parts.py
Function: get_tiles at line 115

Line #      Hits         Time  Per Hit   % Time  Line Contents
==============================================================
   115                                               @profile
   116                                               def get_tiles(self, crop_to=None):
   117        16        100.0      6.2      0.0          stackheight = self.tileshape.nav.size
   118        16        480.0     30.0      0.0          tile_buffer = np.zeros(self.tileshape.flatten_nav(), dtype=self.dataset.dtype)
   119        16       2302.0    143.9      0.1          files_by_tile = list(grouper(self.files, stackheight))
   120        16        118.0      7.4      0.0          detector_size = tuple(self.tileshape.sig)
   121                                           
   122        16         22.0      1.4      0.0          partition_origin = self.slice.origin
   123                                           
   124        16         21.0      1.3      0.0          files_done = 0
[... 137 lines cut ...]

Line #      Hits         Time  Per Hit   % Time  Line Contents
==============================================================
    11                                           @profile
    12                                           def main():
    13         1          5.0      5.0      0.0      ctx = api.Context(executor=InlineJobExecutor())
    14                                           
    15         1          1.0      1.0      0.0      ds = RawFileDataSet(
    16         1          2.0      2.0      0.0          path="/home/clausen/Data/EMPAD/scan_11_x256_y256.raw",
    17         1          2.0      2.0      0.0          scan_size=(256, 256),
    18         1          1.0      1.0      0.0          detector_size_raw=(130, 128),
    19         1          1.0      1.0      0.0          crop_detector_to=(128, 128),
    20         1          1.0      1.0      0.0          tileshape=(1, 8, 128, 128),
    21         1          9.0      9.0      0.0          dtype="float32"
    22                                               )
    23         1          2.0      2.0      0.0      ds.initialize()
    24                                           
    25         1          7.0      7.0      0.0      job = ctx.create_mask_analysis(dataset=ds, factories=[lambda: np.ones(ds.shape.sig)])
    26                                           
    27         1    2094656.0 2094656.0    100.0      result = ctx.run(job)

real	0m2.873s
user	0m2.643s
sys	0m0.214s
```

sk1p, 2018-12-20:
With some optimization, I got the following results (single threaded run, profiler disabled):

Single large file:

```
real	0m1.787s
user	0m1.498s
sys	0m0.270s
```

2^16 small files:

```
real	0m2.844s
user	0m2.143s
sys	0m0.672s
```

So LiberTEM should still be usable for these kinds of datasets, once a proper reader is implemented, even though working on many small files takes about 1.6x as much time. For the use-case of opening simulated datasets, this should not be as pronounced, as files are larger (~1MB) than in my testcase.

uellue, 2018-12-20:
Nice! That looks good.

sk1p, 2018-12-20:
Updated the first comment with the steps necessary to have full support for the "raw file set" format.

For loading [Dr. Probe](https://github.com/ju-bar/drprobe_clt) files in a user-friendly way, some metadata may be necessary, either entered by the user when loading, or loaded from some JSON sidecar file. CC @ju-bar

Also, the output of Dr. Probe has an additional dimension, the sample thickness. At the beginning, we could support loading a single thickness in the UI, later, we could add a navigation slider for the additional dimension, and re-run the job for the selected thickness. From the scripting interface, we can already support the additional dimension; for example using a sparse matrix as mask, which is only populated for the thickness you are interested in.

Another possibility would be to just display the different thickness results one below another, but that only scales to a handful of results.

sk1p, 2023-03-20:
Discussion with @uellue and @matbryan52: this is related to having a descriptor of either the file set, or the whole dataset. For example, one could allow users to create a descriptor for a glob like this:

```python
from libertem.io.base import FileSetDescriptor
fds = FileSetDescriptor.glob("stuff/*.bin")

ds = ctx.load('raw', descriptor=fsd)

# under the hood:
RawFileDataSet(descriptor=fsd, path=None)
```

Or, alternatively, the descriptor could also include information about the data set, which is #1376

uellue, 2025-03-26:
Deferred until real world use case encountered. Dask data sets are an alternative solution.