LiberTEM/LiberTEM #211: Highlight the focus for keyboard controls

uellue, 2018-12-14:
This relates to the keyboard controls in #164 

Right now it can be difficult to tell which element has the focus for keyboard controls.

Perhaps change the pointer or somehow highlight the display that has keyboard focus, just like the text boxes here at GitHub?

sk1p, 2019-01-02:
Agreed - having some kind of outline for focused elements is a best practice. Need to check how to implement this for SVG elements. The default outline is sadly quite ugly...

Quick mockup:

[image]

uellue, 2019-01-06:
Looks good to me!

Here's how Inkscape does it:
[image]

Would it make sense to follow the style of Inkscape, just to have a familiar look?

sk1p, 2019-02-05:
I implemented a first version of this in the complex branch, which will be merged soon. I just change the color when the handle has keyboard focus. I kind of dislike the Inkscape way a bit, because filling the circles will hide what's behind them. Maybe we should wait for some user feedback before making further changes.

closed 2019-02-05 by sk1p

uellue, 2019-02-05:
Looks good! :-)

commit ee5e7d47 (2019-02-05): Re-implemented keyboard focus and control for handles

 + focus indicator, maybe needs more style tweaks, closes #211
 + can focus all handles individually, closes #212
commit cf586cf1 (2019-02-05): Support for complex-valued datasets (#228)

 + all results can now have multiple channels (example: magn, real, imag, ...)
 + com for complex numbers: follow scipy definition
 + GUI: Unify frame and result components
 + decouple ring/disk widgets from the handles and put logic into
   the analysis components
 + move pick handling from Result into its own component, which wraps DraggableHandle
 + include styled-components dependency, replace some bare style props
 + focus indicator, maybe needs more style tweaks, closes #211
 + can focus all handles individually, closes #212