LiberTEM/LiberTEM #218: Event loop was unresponsive in Worker for 3.45s.

sk1p, 2018-12-17:
When connecting to an existing dask cluster, the worker sometimes logs the following message:

```
distributed.core - INFO - Event loop was unresponsive in Worker for 3.43s.  This is often caused by long-running GIL-holding functions or moving large chunks of data. This can cause timeouts and instability.
```

This needs to be debugged and the root cause needs to be found.

sk1p, 2018-12-20:
Observation: this appears to mostly happen for the first operation, maybe this is related to initially importing the required Python modules?

uellue, 2018-12-20:
Yes, this sounds about right. If the workers are started by forking from a local process, they won't have to load modules.

uellue, 2018-12-21:
I'll try the --preload option to see if that helps:

 http://distributed.dask.org/en/latest/setup.html#customizing-initialization

closed 2018-12-21 by sk1p

uellue, 2018-12-21:
Fix confirmed with test on moellenstedt.

PR #223 (2018-12-21): Preloading for some modules
Loading modules as part of job execution can make dask workers
unresponsive for too long. This module allows pre-loading any offenders
at the time a worker starts up.

Usage:
dask-worker [...] --preload libertem.preload [...]

Fixes #218
commit a192d003 (2018-12-21): Preloading for some modules

Loading modules as part of job execution can make dask workers
unresponsive for too long. This module allows pre-loading any offenders
at the time a worker starts up.

Usage:
dask-worker [...] --preload libertem.preload [...]

Fixes #218