LiberTEM/LiberTEM #220: FabIO is a Python module for reading and handling data from two-dimensional X-ray detectors.

uellue, 2018-12-19:
The data formats supported by FabIO are relevant for LiberTEM. Can we use FabIO for reading data? Perhaps also contribute to FabIO?

sk1p, 2019-12-02:
We just had a use case for this: reading and processing data in EDF format.

Generally, it should be possible to create a `DataSet` that consumes a `FabioImage`. You can ask fabio for the number of frames using [`FabioImage.nframes`](https://github.com/silx-kit/fabio/blob/165baf6e54b5dbbc69f3687387b5c91aba6a1005/fabio/fabioimage.py#L440) and get individual frames using [`get_frame(num)`](https://github.com/silx-kit/fabio/blob/165baf6e54b5dbbc69f3687387b5c91aba6a1005/fabio/fabioimage.py#L447).

I think the challenge then is 1) ensuring that this is "lazy" for all fabio formats and 2) mapping the individual images to discrete coordinates as currently used by LiberTEM.

For 1) we could create a whitelist of "well-behaved" formats, and later patch fabio, to prevent memory-related crashes when all libertem workers try to open the whole dataset at the same time...

Regarding 2), this is mostly important for analyses that take the spatial coordinates into account (note that for this kind of analysis, it would be preferable to have basic support for physical units, or even direct access to the metadata stored for each frame at the beginning) - otherwise we can read and process the images in whatever order they are given to us, and the user can re-arrange the result according to some index list. This is of course a challenge in the web interface...

sk1p, 2019-12-02:
> Generally, it should be possible to create a DataSet that consumes a FabioImage. You can ask fabio for the number of frames using FabioImage.nframes and get individual frames using get_frame(num).

This is also what `FileSeries.frames` does - with an optimization for edf using `lazy_iterator`. So `FileSeries` could be extended by `frames_for_range` or similar, which would take care of also supporting something like the `lazy_iterator` of `EdfImage`.

sk1p, 2020-10-15:
/cc @SimeonEhrig