LiberTEM/LiberTEM #221: Support for complex numbers

uellue, 2018-12-20:
@fwin22 would like support for complex number input data in order to directly process simulation results with both amplitude and phase. @fwin22, how are real and imaginary part saved? numpy-compatible, i.e. (re, im), or do we have to assemble the components from hand somehow?

Items to check:
- [ ] Input format, see above
- [ ] Back-end support for complex numbers and switching based on type if necessary: numpy, pytorch scipy.sparse
- [ ] Check for type casts throughout the code
- [ ] Test coverage
- [ ] Analysis with results as complex, im, re, abs, &theta;
- [ ] Visualization similar to COM

FWin22, 2018-12-20:
@uellue either in numpy format ('complex64' or complex'128') or as binary data.

sk1p, 2018-12-20:
If you say numpy format, you mean `.npy` or `.npz` files, right? And binary data is something compatible with `np.fromfile(..)`? For the first, we need to add explicit support, the other only needs some small changes in the raw reader.

uellue, 2018-12-20:
I meant something that can be used with numpy.memmap with "numpy format". :-)

FWin22, 2018-12-20:
yes to both.

uellue, 2018-12-21:
@FWin22 Do you perhaps have a test data set? :-)

sk1p, 2018-12-21:
On the `complex` branch, there is now support for working with complex-valued data. There are still a few items to do, namely:

1) Review/fix calculation for center of mass for complex-valued data. I'm pretty sure this is currently not correct. 
2) `.npy` support - should work pretty much like the raw format, with less parameters to specify. We track this in issue #222 so I'll remove it from the list in the issue description.
3) Multi-channel response for frame picking. This needs user interface changes.

uellue, 2018-12-21:
Looking for references regarding center of mass for complex numbers, the best I could find was the [definition of the center of mass](https://en.wikipedia.org/wiki/Center_of_mass#A_system_of_particles): 
[image]

That calculation works with complex numbers, too. The resulting center of mass is a complex number. Regarding visualization or interpretation of that I don't have a good idea.

@FWin22, would you use a center of mass calculation on your data, and what is the expected outcome if yes? Would you do that directly or rather convert a complex amplitude to a real intensity before?

If we can't find a sensible definition, we should probably raise a `NotImplementedError` rather than just calculating something.

sk1p, 2018-12-21:
> That calculation works with complex numbers, too. The resulting center of mass is a complex number. Regarding visualization or interpretation of that I don't have a good idea.

Yeah, that's what scipy is doing, but I don't think that has a meaningful physical interpretation :grin: I think converting to intensity beforehand is a good idea, as it means the center-of-mass output has the same meaning in complex/real/int/... cases. Maybe, if there is another meaningful way to calculate COM, we could do both for complex data?

uellue, 2019-01-08:
@sk1p @FWin22 just to come back to the discussion regarding center of mass of complex numbers. If we have a complex number data set, that will most likely be amplitude values, correct?

In that case it could be reasonable to simply transform the complex amplitude to a real intensity with |A|^2 or AA* and then process it. Would that be a good behavior for center of mass only, or does that also make sense for virtual detectors?

uellue, 2019-01-08:
@FWin22 would converting to intensity work for you?

closed 2019-02-05 by sk1p

uellue, 2019-02-05:
Works, awesome! 👍

PR #228 (2019-02-05): Support for complex-valued datasets
This adds support for reading and processing complex-valued data, currently only by using the raw DataSet. Closes #221. This is still WIP, see #221 for remaining work.