LiberTEM/LiberTEM #234: Correct bounds check for frame picking

uellue, 2019-01-08:
How to reproduce: Use pick mode, preferably on a K2IS data set with low scan resolution that throws an exception when reading out of bounds.

Observed behavior: With the mouse, you can move the picker to the bottom and right border. This is an off-by-one pick because the picker sits in the left upper corner of a pixel. With the keyboard, you can even move the picker out of the frame.

Expected behavior: Correct bounds check for mouse and keyboard control. Ideally, the picker should sit in the middle of a pixel (enhancement).

sk1p, 2019-01-22:
The mouse movement should now be properly bounded by the containing image, keyboard and detailed positioning TBD (probably together with #212, #211 and related issues).

sk1p, 2019-02-05:
Keyboard movement should now be properly constrained to the containing image, since cf586cf1f1b2bb42488419ea9f4cfb1a5a6d0f02

closed 2019-02-05 by sk1p

uellue, 2019-02-05:
Perfect! 💯

commit 1ac70c6e (2019-02-05): GUI: center handles on their frames, closes #234