LiberTEM/LiberTEM #243: Support for ser/emi files (TIA)

uellue, 2019-01-11:
Support for ser/emi is required for a specific project. 

http://www.er-c.org/cbb/info/TIAformat/
http://www.er-c.org/cbb/info/TIAformat/TIAseriesformat.pdf
http://hyperspy.org/hyperspy-doc/current/user_guide/io.html#fei-tia-ser-and-emi
https://ercius.wordpress.com/2014/01/21/electron-microscopy-data-files-emi-ser-update/
https://imagej.nih.gov/ij/plugins/tia-reader.html

@ercius https://github.com/ercius/openNCEM/blob/master/ncempy/io/ser.py that is the reader you wrote? Any chance that you could re-license it under a more permissive open source license? :-)

ercius, 2019-01-12:
Yes, that's our reader! :)

Changing the license is on my list of things to do. I'm putting together a
new release with a few updates and bug fixes. The new license will be
included. I'll do it by next week. Thanks for your patience.

This will 100% happen. You can start using it if you need it sooner.

uellue, 2019-01-14:
@ercius 👍 👍 👍  Great, thank you! :-)

uellue, 2019-01-17:
@ercius 🎉 🍾  Thank you! @sk1p just said he'll get started with that today. 👍

sk1p, 2019-01-21:
Initial support for SER/EMI files via ncempy is now available in the Python API, thanks again @ercius! I still need to add the GUI part, write integration tests and do a performance analysis. As I only have 3D datasets available (line scans), which are not yet really suported in the GUI, I'll postpone the GUI part for now. I've updated the issue description with the items still TODO.

@uellue for whatever reason, the link checker doesn't like the ER-C link, which makes the travis build fail. Any idea about that?

uellue, 2020-01-23:
Wasn't that implemented in #243? Can we close this?

sk1p, 2020-01-28:
So, the two unchecked boxes are still relevant: there are no integration tests yet, and I don't know about the performance - I haven't seen any large sample files yet. The second point is optional, I'd say (if no one screams...), the first one still needs to be done.

uellue, 2021-09-06:
We do run this in CI and performance hasn't really been an issue so far, as far as I am aware. Closing!

closed 2021-09-06 by uellue