LiberTEM/LiberTEM #256: Handle missing data

sk1p, 2019-01-28:
Sometimes, the acquisition may be canceled and the dataset may not contain as many images to fill the whole scan region. We should handle this case in some way. Related to #120 and maybe also our live acquisition plans.

In the FRMS6 format, the "signalFrames" header value will contain the real number of acquired frames. Maybe we can just ignore that we don't have data available, output a warning etc.

uellue, 2019-01-28:
What about displaying the missing data as black or checkerboard?

sk1p, 2019-01-29:
Black could work, as long as there is no need to distinguish between 0 and "no data". Checkerboard could be problematic for missing values that are more spread around, like randomly missing frames. I guess we should have general visualization settings, where the user can select the appropriate placeholder (could also be any color that is not part of and very different from the current colormap).

Previous thoughts on this, from #128:

> 
> How do we, generally, handle "missing" or invalid values in visualization? Examples:
> 
>  - complete frames may be missing, for example because the acquisition did not keep up (with UDP used in acquisition this is a real-world scenario!)
>  - values may fall out of the range of visualization functions, for example negative numbers with log scaling
>  - other examples?
> 
> For specific cases, we should try to find a good place holder (as with center of mass: the origin / reference center), but if we have real missing data, this is not so easy. Maybe a bright neon colored marker? Like [darktable clipping indicator](https://redmine.darktable.org/attachments/download/2582/overexposed-toggle_on.png) or for larger areas, [darktable raw over exposure indication](https://www.darktable.org/2016/10/raw-overexposed/rawoverexposed-2-1.jpeg)

sk1p, 2019-06-12:
This can now also come up if we visualize a result from running a UDF with a `roi` set. In that case, values masked out are `NaN` in the result. So, I propose:

 * augment `visualize_simple` to properly handle `NaNs`
 * add a method to query the dataset for missing data, which returns a bool mask array
 * as a postprocessing step, add `NaNs` in the places of missing data (udf: `run_for_dataset*`, for the old-style built-in jobs we may have to do this in each `Job`)

PR #793 (2020-12-14): sync_offset, reshape datasets and get coords for UDFs
## Google Summer of Code '20 project

Fixes #256, fixes #553 and fixes #726 and

* The Python API and GUI now allows specifying `nav_shape` and `sig_shape` parameters to set a different shape
* `scan_size` and `detector_size` parameters are deprecated in favor of `nav_shape` and `sig_shape` respectively
* All datasets handle missing data gracefully
* The Python API and GUI now allow specifying a `sync_offset` to handle synchronization/acquisition problems
* Users can now access the coordinates of a tile/partition slice through UDFMeta.coord [... cut]
closed 2020-12-14 by sk1p