LiberTEM/LiberTEM #258: Handle `disable_quickedit` failure gracefully

sk1p, 2019-01-28:
When using the windows git bash with the alternative tty thingy (mintty I think?), disabling quick edit mode fails loudly (ask A.Krings for specific error msg), because while we are on windows, the different console doesn't support the whole console mode stuff. Ideally the exception should be caught, and maybe a warning should be logged stating that `disable_quickedit` failed.

Akhileshpm, 2020-02-17:
can you explain a little more about this issue because I am a beginner .

sk1p, 2020-02-17:
As far I remember, the steps to reproduce were as follows:

1) install windows git bash with mintty
2) run `libertem-server` from a windows bash window

That should complain loudly with some windows-specific error message. Let me know if you can reproduce this issue or need more help!

Akhileshpm, 2020-02-18:
may i know how to run the 2nd step.

sk1p, 2020-02-18:
Hi @Akhileshpm ,

> may i know how to run the 2nd step.

Please follow the installation instructions from https://libertem.github.io/LiberTEM/install.html, then start the libertem server as described in the usage documentation: https://libertem.github.io/LiberTEM/usage.html#starting-the-libertem-server

To reproduce this bug specifically, make sure to run `libertem-server` in a "bash for windows" window configured for using mintty.

sayandip18, 2020-03-31:
I see this issue is more than year old, is it fixed yet?

sk1p, 2020-04-01:
> I see this issue is more than year old, is it fixed yet?

I don't think it is fixed, can you reproduce it?

sayandip18, 2020-04-01:
I think so. This is what I got.
```
Traceback (most recent call last):
  File "c:\users\sandip\python\Lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\sandip\python\Lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\Sandip\.virtualenvs\libertem258\Scripts\libertem-server.exe\__main__.py", line 7, in <module>
  File "c:\users\sandip\.virtualenvs\libertem258\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\sandip\.virtualenvs\libertem258\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "c:\users\sandip\.virtualenvs\libertem258\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\sandip\.virtualenvs\libertem258\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "c:\users\sandip\.virtualenvs\libertem258\lib\site-packages\libertem\web\cli.py", line 23, in main
    console_tweaks()
  File "c:\users\sandip\.virtualenvs\libertem258\lib\site-packages\libertem\cli_tweaks.py", line 7, in console_tweaks
    disable_quickedit()
  File "c:\users\sandip\.virtualenvs\libertem258\lib\site-packages\libertem\win_tweaks.py", line 50, in disable_quickedit
    mode = get_console_mode()
  File "c:\users\sandip\.virtualenvs\libertem258\lib\site-packages\libertem\win_tweaks.py", line 34, in get_console_mode
    raise ctypes.WinError(err)
OSError: [WinError 0] The handle is invalid.

```

sayandip18, 2020-04-01:
Also, [this](https://medium.com/@presh_onyee/activating-virtualenv-on-windows-using-git-bash-python-3-7-1-6b4b21640368) this helped me and I think we can include this in the documentation maybe after fixing the issue.

sk1p, 2020-04-01:
> I think so. This is what I got.

:+1: - looks like it!

>  I think we can include this in the documentation

Sure, maybe as a link in a "notes" section at the bottom of the installation page?

sayandip18, 2020-04-01:
>
> 
> Sure, maybe as a link in a "notes" section at the bottom of the installation page?

Ok!

sayandip18, 2020-04-03:
I just came across the [system error codes](https://docs.microsoft.com/en-us/windows/win32/debug/system-error-codes--0-499-). Here, code 0 actually denotes successful completion of operation whereas I've got WinError 0. How is this possible?

sayandip18, 2020-04-03:
Also, invalid handle is supposed to be error code 6.

sk1p, 2020-04-03:
> So, do I start a PR? I'll make more changes as things get clearer

Sure thing!

sayandip18, 2020-04-15:
I did this 
`try: 
    if getConsoleMode(file_handle, byref(mode)):
            return mode.value
    except ctypes.WinError(get_last_error()): 
            print("FAILED!")`

in line 32, 33 of `win_tweaks.py` and got this as error msg

`$ libertem-server
Traceback (most recent call last):
  File "C:\Users\SANDIP\virenv\lib\Scripts\libertem-server-script.py", line 11, in <module>
    load_entry_point('libertem', 'console_scripts', 'libertem-server')()
  File "c:\users\sandip\virenv\lib\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\sandip\virenv\lib\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "c:\users\sandip\virenv\lib\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\sandip\virenv\lib\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "c:\users\sandip\libertem\src\libertem\web\cli.py", line 18, in main
    console_tweaks()
  File "c:\users\sandip\libertem\src\libertem\cli_tweaks.py", line 7, in console_tweaks
    disable_quickedit()
  File "c:\users\sandip\libertem\src\libertem\win_tweaks.py", line 51, in disable_quickedit
    mode &= ~ENABLE_QUICK_EDIT
TypeError: unsupported operand type(s) for &=: 'NoneType' and 'int'
`

sayandip18, 2020-04-16:
After logging both `disable_quickedit()` and `get_console_mode()` I got a completely different error message

`$ libertem-server
[2020-04-16 12:16:39,310] INFO [libertem.web.server.main:127] listening on localhost:9000
Traceback (most recent call last):
  File "C:\Users\SANDIP\virenv\lib\Scripts\libertem-server-script.py", line 11, in <module>
    load_entry_point('libertem', 'console_scripts', 'libertem-server')()
  File "c:\users\sandip\virenv\lib\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\sandip\virenv\lib\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "c:\users\sandip\virenv\lib\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\sandip\virenv\lib\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "c:\users\sandip\libertem\src\libertem\web\cli.py", line 19, in main
    run(host, port, browser, local_directory)
  File "c:\users\sandip\libertem\src\libertem\web\server.py", line 139, in run
    main(host, port, event_registry, shared_state)
  File "c:\users\sandip\libertem\src\libertem\web\server.py", line 129, in main
    app.listen(address=host, port=port)
  File "c:\users\sandip\virenv\lib\lib\site-packages\tornado\web.py", line 2116, in listen
    server.listen(port, address)
  File "c:\users\sandip\virenv\lib\lib\site-packages\tornado\tcpserver.py", line 151, in listen
    sockets = bind_sockets(port, address=address)
  File "c:\users\sandip\virenv\lib\lib\site-packages\tornado\netutil.py", line 174, in bind_sockets
    sock.bind(sockaddr)
OSError: [WinError 10048] Only one usage of each socket address (protocol/network address/port) is normally permitted
`

sk1p, 2020-04-16:
> TypeError: unsupported operand type(s) for &=: 'NoneType' and 'int'

It seems `mode` is `None` in that expression - you can't do bitwise operations with `None`. I think you need to try to find the right part of the code to catch the error - if you just return `None` from `get_console_mode`, any user of that function will need to check for `None`.

sayandip18, 2020-04-16:
I meant modifying the functions to catch the exception. Also, I think I have managed to fix the issue. Will start a PR soon.

sk1p, 2020-04-30:
Fixed by #731

closed 2020-04-30 by sk1p