LiberTEM/LiberTEM #261: Separate LiberTEM into sub-modules that can be installed and used individually

uellue, 2019-02-03:
Discussion with @sk1p:

Now that we are starting to integrate application modules, we should make sure these are properly packaged and can be installed separately from each other to preempt https://github.com/hyperspy/hyperspy/issues/1599. See also #259!

With independent modules we isolate the core of LiberTEM from proliferation of dependencies. In particular application modules with dependency on hyperspy have to be isolated, because hyperspy depends on a larger part of the pydata universe. Similarly, application modules might be useful without the rest of LiberTEM and its dependencies.

Furthermore, users might want to integrate application code that is covered by third-party patents. Here we should make it possible to install a "clean" version of LiberTEM without patented applications, in particular for redistribution by vendors or use outside of research.

The following components could make proper independent items that can be used separately:

libertem-core: Processing back-end
libertem-server: Web interface back-end and front-end
libertem-api: (Perhaps later) client-side API for libertem-core under MIT license
libertem-io: Low-level high-performance IO libraries for various file formats. Can be maintained by LiberTEM (most notably K2IS and MIB reader where @sk1p did substantial work) or interface to third-party code like https://github.com/ercius/openNCEM/tree/master/ncempy by @ercius for ser and dm4 support.
libertem-xyz: Application-specific sub-module xyz

Question: Would it make sense to split up the IO module into modules for individual file formats?

Higher-level modules:
libertem-all: all
libertem-base: libertem-core, libertem-server, libertem-api, libertem-io, libertem-mask (virtual detector application)

Each application-specific module should have its own DOI to give proper credit and make it citable. That can be a new one with upload on zenodo.org if the main application code is part of libertem (example: Karina's code), or an existing DOI for modules that mostly integrate third-party application code.

An application-specific module can contain a combination of processing facilities that use libertem-core and are exposed via libertem-api together with web interface back-end and front-end code. As a consequence, we need an extension model for API and web interface.

LiberTEM should be built from a single code base for the time being to ensure consistency and integration testing of all the parts when we make changes to the interfaces. When interfaces have stabilized, this could be separated into individual code bases.

We should evaluate the impact on automated tests because they could take longer if we have to set up test environments with the better half of pydata. If that becomes an issue, we could consider compartmentalizing the tests somehow.

We have to decide if we release packages in lock-step or if we allow separate release cycles. For the time being we likely have to release in lock-step until interfaces have stabilized to ensure that a `pip install libertem-all` always installs a working, coherent setup with the most recent release.

dnjohnstone, 2019-02-03:
Sounds like a great idea to me also with respect to our pyxem conversation!

sk1p, 2020-01-27:
We also need to think about the Python package namespace structure. The way I see it, there are two possibilities:

 1) Each package lives in its own namespace (for example, `libertem-core` in `libertem`, `libertem-gridmatcher` in `libertem_gridmatcher`, `libertem_blobfinder`, ...)
 2) The packages share the `libertem` namespace, for example by putting all external packages into `libertem.ext` or similar: `libertem.ext.gridmatcher`, `libertem.ext.blobfinder`, ...

The first one is easiest to implement, but not so pretty. The second one requires that we convert libertem and all "extension packages" into [namespaces packages](https://packaging.python.org/guides/packaging-namespace-packages/). Among other things, that means we may need to remove the `__init__.py` from `src/libertem/`. Currently, it holds versioning information; maybe we can move that to `libertem/core/__init__.py` or similar? Need to study [PEP420](https://www.python.org/dev/peps/pep-0420/) and experiment a bit - maybe we can also just have `libertem.ext` as the "extension point" package by not having `libertem/ext/__init_.py`?

uellue, 2020-01-27:
Actually, this brings questions about the relation between the components and potential users of them. Some parts in the current code base are rather general and mathematical, such as gridmatching, masks¹ and some high-performance numerics code like correlation in blobfinder. They could, in principle, be spun off into their own project or merged into another one since they are perfectly usable without any other part of LiberTEM. Here, the underscore would feel right to me. Essentially, that is just branding, right? It could be called `asdf_ghjk` for all it matters?

Then we have UDFs that are currently only useful together with LiberTEM core. Some of the UDFs are core components such as auto, sum, masks and pick. Others are application-specific like the blobfinder, FEM, .... Here we could have components that we have to isolate from the rest because of patent questions, by the way. That means here the underscore is probably the better choice as well? Ptychography UDFs should probably be branded with Ptychography 4.0 instead of LiberTEM anyway?

As we discussed before, the Analysis classes are  a bit more tricky since they currently rely on internal APIs to provide data for the web interface that WILL be subject to change as part of #134. This will need more time to stabilize. Currently, the Analyses are required for the web interface and for the API.

Looking at the API, we could perhaps move the methods in `api.Context` to create Analyses elsewhere or use the constructor of Analysis classes directly like we do with UDFs? That way, we don't have a central point that requires ALL Analyses. The methods are just wrappers that create the `parameters` dict from method arguments and don't rely on anything else in the Context. Plus, we have several Analyses that are not even present in `api.Context` and are only available through the GUI and their (undocumented?) constructor.

In summary, we could start by spinning out `blobfinder` as `libertem_blobfinder`? That can then be a template to spin out components like `libertem_masks`, `libertem_gridmatching` etc. This one doesn't have an associated Analysis, so we can leave answering this question for later.

The Analysis classes would be candidates for `libertem.analysis.*` namespace packages at a later time since they are closely linked with LiberTEM core and form one product?

¹ *As a side note, we could do a nice refactor of the mask infrastructure with a class-based interface, like the `ProtoTileStream` concept?*

sk1p, 2020-01-28:
> Actually, this brings questions about the relation between the components and potential users of them. Some parts in the current code base are rather general and mathematical, such as gridmatching, [...]

True.

> Then we have UDFs that are currently only useful together with LiberTEM core. Some of the UDFs are core components such as auto, sum, masks and pick.

So these should stay in the core package, right?

> Others are application-specific like the blobfinder, FEM, .... Here we could have components that we have to isolate from the rest because of patent questions, by the way. That means here the [...]

True about the branding and patents. So the structure(s) would be like this:

1) Complex applications, for example blobfinder: three parts: LiberTEM core, blobfinder UDF, blobfinder "core" functions (could even be split up further, into correlation and application-specific post-processing functions). Only the blobfinder UDF package depends on LiberTEM, the other parts are independent and reusable.

2) Simple applications, for example FEM: start having a single integrated package, which contains both the UDF and the application-specific functions/classes/...

> As we discussed before, the Analysis classes are a bit more tricky since they currently rely on internal APIs to provide data for the web interface that WILL be subject to change as part of #134. [...]

Maybe we should rethink the `Analysis` class from the ground. It was built for the web API, but then also used for the `Context` API. I think it would be a good idea to think about splitting these two concerns - and I agree with moving the entry point for analyses away from `Context`. I'm not sure yet what the best way is going forward - I need to think a bit on this.

> Looking at the API, we could perhaps move the methods in `api.Context` to create Analyses elsewhere or use the constructor of Analysis classes directly like we do with UDFs? That way, we don't have [...]

> In summary, we could start by spinning out `blobfinder` as `libertem_blobfinder`? That can then be a template to spin out components like `libertem_masks`, `libertem_gridmatching` etc. This one doesn't have an associated Analysis, so we can leave answering this question for later.

I would go a bit further and say that we should have one (or more) template repositories if we want to continue splitting of packages - creating a new package should consist mostly of instantiating this repository and setting up integrations (CI, PyPi, ...). See also: [cookiecutter](https://cookiecutter.readthedocs.io/en/1.7.0/).

> The Analysis classes would be candidates for `libertem.analysis.*` namespace packages at a later time since they are closely linked with LiberTEM core and form one product?

Hmm, possible, but maybe not necessary - I guess for the web API we will need some kind of registry, which can be implemented independent of namespacing.

> ¹ _As a side note, we could do a nice refactor of the mask infrastructure with a class-based interface, like the `ProtoTileStream` concept?_

Hmm, can you provide an example? We should also keep in mind not to make the API too complicated...

uellue, 2020-01-28:
> * Complex applications, for example blobfinder: three parts: LiberTEM core, blobfinder UDF, blobfinder "core" functions (could even be split up further, into correlation and application-specific post-processing functions). Only the blobfinder UDF package depends on LiberTEM, the other parts are independent and reusable.
> * Simple applications, for example FEM: start having a single integrated package, which contains both the UDF and the application-specific functions/classes/...

Yes, this sounds perfect! :-)

uellue, 2020-01-29:
> > ¹ _As a side note, we could do a nice refactor of the mask infrastructure with a class-based interface, like the `ProtoTileStream` concept?_
> 
> Hmm, can you provide an example? We should also keep in mind not to make the API too complicated...

Many of the functions share the same parameters: dimensions, use_sparse, center points, radii. Furthermore, there's the aspect of moving from using the coordinates directly to generating coordinate maps and then applying filters to them. Nothing urgent for now. We can maybe have a look at it when we are actually splitting this part off the main repository?

uellue, 2020-01-29:
> So these should stay in the core package, right?

Yes, I'd say so, at least for the time being. :-)

uellue, 2020-01-29:
About the documentation: Probably all the `libertem_*` branded packages can be documented in the main LiberTEM documentation, with a proper overview of the subpackages so that people know what to install and import from where?

uellue, 2020-01-29:
> I would go a bit further and say that we should have one (or more) template repositories if we want to continue splitting of packages - creating a new package should consist mostly of instantiating [...]

👍 for template repositories. Blobfinder can be the testing ground for that, and then rinse and repeat, bit by bit. :-)

uellue, 2020-01-29:
> Hmm, possible, but maybe not necessary - I guess for the web API we will need some kind of registry, which can be implemented independent of namespacing.

Ok, probably we can put the web interface and Analysis aspect on the "maybe later" pile. Probably we will have a better idea once we have tackled the easier parts such as UDFs and generic code successfully.

sk1p, 2020-02-04:
Initial version of template repository is done:

https://github.com/LiberTEM/template

can be tested via:

```
$ pip install cookiecutter
cookiecutter https://github.com/LiberTEM/template
```

and giving some values for variables.

sk1p, 2020-02-05:
Current progress of `libertem_blobfinder` extraction: https://github.com/LiberTEM/LiberTEM-blobfinder/milestone/1

sk1p, 2020-02-12:
https://pypi.org/project/libertem-blobfinder/ v0.4.0 is now released

uellue, 2021-05-27:
This is an ongoing task. Step by step!