LiberTEM/LiberTEM #262: Proper way to do type checks with user-defined functions

uellue, 2019-02-03:
The result buffers for user-defined functions are declared with specific dtypes. Apparently numpy accepts unsafe casts when assigning values to these buffers without warning, by default. As an example, assigning a float32 to a u2 buffer just silently truncates the value and/or creates an overflow.

```
In [1]: import numpy as np

In [2]: a = np.array([1, 2, 3], dtype='u2')

In [3]: a[0] = 0.1

In [4]: a[0] = -10.1

In [5]: print(a[0])
65526
```

We need a fast and convenient way to catch such assignments because they are programming errors in 99 % of the cases.

sk1p, 2019-02-09:
A possibility is a type checking proxy object around the buffer arrays. Quick prototype using [`wrapt`](https://pypi.org/project/wrapt/):

```
In [1]: class TypeCheckedArray(wrapt.ObjectProxy): 
    ...:     def __setitem__(self, k, v): 
    ...:         if not isinstance(v, np.ndarray): 
    ...:             v = np.array(v) 
    ...:         if not np.can_cast(v.dtype, self.__wrapped__.dtype, casting='safe'): 
    ...:             raise TypeError("can't safely cast from {} to {}".format(v.dtype, self.__wrapped__.dtype)) 
    ...:         self.__wrapped__[k] = v 
    ...:                                                                                                                         

In [2]: a = TypeCheckedArray(np.array([1, 2, 3], dtype='u2'))                                                                   

In [3]: a[0] = 0.1                                                                                                              
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-3-5338e63b4d85> in <module>
----> 1 a[0] = 0.1

<ipython-input-1-9ddea37cfcc2> in __setitem__(self, k, v)
      4             v = np.array(v)
      5         if not np.can_cast(v.dtype, self.__wrapped__.dtype, casting='safe'):
----> 6             raise TypeError("can't safely cast from {} to {}".format(v.dtype, self.__wrapped__.dtype))
      7         self.__wrapped__[k] = v
      8 

TypeError: can't safely cast from float64 to uint16

In [4]: a[0] = np.array(42, dtype="u2")                                                                                         
```

Looks good so far, but then also requires explicit construction of an ndarray, because `np.array` constructs 64 bit types by default:

```
In [5]: a[0] = 42                                                                                                                                                
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-5-1648fad9c8fa> in <module>
----> 1 a[0] = 42

<ipython-input-1-9ddea37cfcc2> in __setitem__(self, k, v)
      4             v = np.array(v)
      5         if not np.can_cast(v.dtype, self.__wrapped__.dtype, casting='safe'):
----> 6             raise TypeError("can't safely cast from {} to {}".format(v.dtype, self.__wrapped__.dtype))
      7         self.__wrapped__[k] = v
      8 

TypeError: can't safely cast from int64 to uint16
```

uellue, 2019-02-25:
`a[0] = np.uint16(42)` should work, right? I think that is tolerable to write. I do like the wrapper solution and much prefer it to the default behavior in this case.

uellue, 2019-02-27:
Hm, this looks somewhat related: http://docs.astropy.org/en/stable/api/astropy.units.quantity_input.html

Perhaps that could be useful as well?

sk1p, 2019-04-03:
In 426f5a59bc310c447d09f9d1d50275dd92938b4d I made a related change for preventing changing the `dest` dict in the merge function instead of the buffer itself. That's also an easy mistake to make, but it's equally easy to prevent here.

uellue, 2025-03-27:
Seems to be partially fixed in current versions (Python 3.13.2, NumPy 2.1.3):

```python
>>> import numpy as np
>>> a = np.array([1, 2, 3], dtype='u2')
>>> a[0] = 0.1
>>> a[0] = -10.1
Traceback (most recent call last):
  File "<python-input-3>", line 1, in <module>
    a[0] = -10.1
    ~^^^
OverflowError: Python integer -10 out of bounds for uint16
>>> print(a[0])
0
```

The truncation is still not great, though.