LiberTEM/LiberTEM #271: Issues with non-trivial picking

uellue, 2019-02-22:
@kruzaeva  and I ran into issues when trying to extract non-trivial data portions with picking. See code example below! I'd expect all examples to work, following the documentation.

Curiously, these errors are not triggered in test_analysis_raw.py, although it contains tests that should cover these cases, as far as I can see. Probably we should have a look at including tests that trigger the behavior here to prevent it from reoccurring, once we know why that happens?

```python
import os
os.environ["OMP_NUM_THREADS"] = "1"
os.environ["MKL_NUM_THREADS"] = "1"
os.environ["OPENBLAS_NUM_THREADS"] = "1"

from libertem import api

if __name__ == '__main__':
    ctx = api.Context()
    
    ds = ctx.load(
        "raw",
        path="C:/Users/weber/Nextcloud/Projects/Open Pixelated STEM framework/Data/EMPAD/scan_11_x256_y256.raw",
        dtype="float32",
        scan_size=(256, 256),
        detector_size_raw=(130, 128),
        crop_detector_to=(128, 128),
    )
    print("DS 1: ", ds.shape)

    ds2 = ctx.load(
        "mib",
        path="C:/Users/weber/Nextcloud/Projects/Open Pixelated STEM framework/Data/data_Knut/MOSFET/20181119 185412/default.hdr",
        scan_size=(256, 256),
        tileshape=(1, 8, 256, 256)
    )
    print("DS 2: ", ds2.shape)

    # works
    pick_job = ctx.create_pick_job(dataset=ds, origin=(0, 0, 0, 0), shape=(1, 1, 128, 128))
    pick_result = ctx.run(pick_job)

    # works
    pick_job2 = ctx.create_pick_job(dataset=ds2, origin=(0, 0), shape=(1, 1, 256, 256))
    pick_result2 = ctx.run(pick_job2)

    # breaks on execution
    pick_job = ctx.create_pick_job(dataset=ds, origin=(0, 0, 0, 0), shape=(1, 2, 128, 128))
    pick_result = ctx.run(pick_job)
    r'''
    ---------------------------------------------------------------------------
    ValueError                                Traceback (most recent call last)
    <ipython-input-48-29b48ddb4fb3> in <module>
    ----> 1 pick_result = ctx.run(pick_job)
        2 print(pick_result)

    c:\users\weber\documents\libertem\libertem\src\libertem\api.py in run(self, job)
        382         for tiles in self.executor.run_job(job_to_run):
        383             for tile in tiles:
    --> 384                 tile.reduce_into_result(out)
        385         if analysis is not None:
        386             return analysis.get_results(out)

    c:\users\weber\documents\libertem\libertem\src\libertem\job\raw.py in reduce_into_result(self, result)
        53 
        54     def reduce_into_result(self, result):
    ---> 55         out = result.reshape(self.data.shape)
        56         out += self.data
        57         return result

    ValueError: cannot reshape array of size 16384 into shape (1,2,128,128)
    '''

    # breaks on creation
    pick_job2 = ctx.create_pick_job(dataset=ds2, origin=(0, 0, 0, 0), shape=(1, 1, 256, 256))
    pick_result2 = ctx.run(pick_job2)
    print(pick_result2)
    r'''
    ---------------------------------------------------------------------------
    ValueError                                Traceback (most recent call last)
    <ipython-input-10-468cfb6bf5fd> in <module>
    ----> 1 pick_job2 = ctx.create_pick_job(dataset=ds2, origin=(0, 0, 0, 0), shape=(1, 1, 256, 256))
        2 pick_result2 = ctx.run(pick_job2)
        3 print(pick_result2)

    c:\users\weber\documents\libertem\libertem\src\libertem\api.py in create_pick_job(self, dataset, origin, shape)
        318             shape = shape.flatten_nav()
        319         slice_ = Slice(origin=origin,
    --> 320                        shape=Shape(shape, sig_dims=raw_shape.sig.dims))
        321         return PickFrameJob(
        322             dataset=dataset,

    c:\users\weber\documents\libertem\libertem\src\libertem\common\slice.py in __init__(self, origin, shape)
        24                 ("cannot build slice with dimensionality of shape/origin mismatch (%d vs %d); "
        25                  "origin=%r, shape=%r") % (
    ---> 26                      len(self.origin), len(self.shape), self.origin, self.shape,
        27                  )
        28             )

    ValueError: cannot build slice with dimensionality of shape/origin mismatch (1 vs 3); origin=(0,), shape=(1, 256, 256)
    '''
```

sk1p, 2019-03-25:
The first appears to be a bug with `PickFrameJob`, currently working on a fix.

The second part of the issue is actually expected, as it is a side-effect of the pick **job** always using the underlying representation (`raw_shape`) of the dataset, which is 3D for MIB. Opening a new issue for the general fix here.

closed 2019-03-25 by sk1p

uellue, 2019-03-25:
@sk1p Fix confirmed for the first error. 👍 Thx!

uellue, 2019-03-25:
The issue for the general fix would be #307 ?

sk1p, 2019-03-25:
Yep!

commit 12b91238 (2019-03-25): fix PickFrameJob(..., squeeze=True), closes #271