LiberTEM/LiberTEM #272: Fourier space dark field imaging

uellue, 2019-02-22:
It would be great to implement Fourier space dark field imaging like @kruzaeva uses for heterogeneity visualization in the GUI. We can use the usual masks -- disk, ring, point. That feature should include Fourier space average and pick mode for the frames.

sk1p, 2019-02-26:
Agreed, this would be a nice feature! I wonder if we should implement this as a derived dataset (and implement #201 while we're at it), and just allow all existing analyses of the GUI for it, or if this should be a new separate analysis.

**Derived**: all existing analyses just work as they are (if they can work with complex data, that is). Generic functionality. Nice performance boost if we implement "materialized" derived datasets. 

We should also think about the GUI workflow. How does the user specify that he wants a derived dataset? Example 1: open dataset (as usual), User clicks on new "button" derived/transformed dataset, can optionally select parameters, ... Example 2: specify transformations when opening the dataset. Users can add/re-order transformations and enter parameters for each of them.

**Analysis**: easier to implement in the backend (don't need to implement derived datasets), but will need changes to replace pick mode with custom results. Will need also need some GUI work. If we want to support disk/ring/point in one analysis, it may also make sense to combine the existing analyses into one (let's call it "virtual detector"), and have a single mask editor that is used for both cases (to keep things consistent).

sk1p, 2019-02-26:
Another thing to consider: do users want to compare fourier-space frames with real-space frames? That wouldn't fit directly into the derived dataset usecase, but rather a "compare datasets at same coordinates" functionality (possibly with prior alignment, then it becomes useful for a broader range of applications, like comparing two 4D datasets)

uellue, 2019-02-27:
> Another thing to consider: do users want to compare fourier-space frames with real-space frames? That wouldn't fit directly into the derived dataset usecase, but rather a "compare datasets at same [...]

Hm, at the moment I am not aware of a specific application for such a view. Let's keep it on the idea list for now until we have a specific need?

sk1p, 2019-06-17:
Just to keep the issue updated: this is now worked on by @kruzaeva

sk1p, 2019-06-26:
See PR #378. Not included yet:

> We can use the usual masks -- disk, ring, point

Only ring implemented for now. @kruzaeva maybe other mask types can be integrated at the end of GSOC, if there is time left?

uellue, 2020-01-23:
Closing this for now since ring seems to be the most useful mask. We can open issues to allow other shapes when we have a specific need.

closed 2020-01-23 by uellue