LiberTEM/LiberTEM #275: Use feature tracking for image stack alignment

uellue, 2019-02-27:
The blob tracking algorithm by @kruzaeva can probably be used to calculate image alignment vectors. This would be an important feature, in particular for in-situ studies.

In a way, it can already track the zero-order peak.

To make it useful in practice, the aligned stack could maybe be made available with #201?

sk1p, 2019-03-27:
Also relevant: https://github.com/bsavitzky/rigidRegistration and related paper

uellue, 2025-03-27:
Related to image stack alignment in LiberTEM-holo. For that application it might be interesting to implement "divide and conquer" distributed stack alignment in case there are many, many images in a stack? Mathematically it should be possible to align within a partition, and then connect the partitions in the merge function.