LiberTEM/LiberTEM #283: Spurious test failure

sk1p, 2019-03-03:
See https://travis-ci.org/LiberTEM/LiberTEM/jobs/501225151 for details

sk1p, 2019-03-03:
As we are using randomly generated inputs, we should seed the random generator, either with a static value (easy debugging) or with a new value for each run (possibly catches more bugs, like in this instance?)

For example, we can set the seed in a fixture like this:

```
rand.seed(0)
np.random.seed(0)
```

And log the seed to the console via [pytest live logging](https://docs.pytest.org/en/latest/logging.html#live-logs) in case of non-static seed.

sk1p, 2019-03-04:
Sorry, hitting the rebuild button actually killed the old log. It was one of the mask application tests failing. I could not yet reproduce this locally.

uellue, 2019-03-04:
Hm, that's an interesting one! Should we attribute it to cosmic rays for now, until it happens again and we find out how to reproduce it?

sk1p, 2019-03-04:
I guess. If we find the time we can add the seed output as described above, so we can actually reproduce it the next time it happens :grin:

PR #446 (2019-10-07): Fixed a glitch where negative peak elevations were possible
Edit: No peak_local_max involved, root cause not sure -- numerical error?

> Peak_local_max didn't give the maximum of the corrmap in some border cases,
> likely in a result that is near zero.

Include more fuzzing and better reporting in test

Closes #283
closed 2019-10-07 by sk1p