LiberTEM/LiberTEM #286: User-defined functions

sk1p, 2019-03-07:
This is an overview issue for user-defined functions. As a first step, we want to support mapping functions over frames. This is already partially done in the `phase-correlation-proto` branch, where we developed this for supporting the analysis developed by @kruzaeva. We still have some work to do to develop a stable and user-friendly interface, and to support some more use-cases:

- [ ] Support for auxiliary data, for example parameters for each frame
- [ ] Integration into the Web API, for showing partial results. See #216 
- [ ] Type-checking for better developer experience: #262 
- [ ] Support for running functions in the cluster without reading an input dataset
- [ ] Support for creating a new dataset, for example by running a distributed simulation

sk1p, 2019-03-21:
UDFs have landed in `master`, at least the low-level functionality. The interface is quite nice already, I think, but may change in the near future, as we implement some more functionality with this.

uellue, 2019-04-02:
Using `BufferWrapper` with `dtype="object"` is confirmed to work. That means we can use pretty much any data type(s) we like as return values for UDFs. Performance TBD!

sk1p, 2019-05-14:
See also the interface discussions in #310 - I'm going to experiment a bit with the interfaces and see if I can come up with something elegant that matches the constraints discussed here and in related tickets.

sk1p, 2019-05-14:
For reference, here is the class-based interface proposed in #310:

```python
class MyUDF_1:
    wish = "floats"
    
    def __init__(self):
        self.local_variable = 42
        
    
    def process(self, frame):
        print(self.local_variable, "UDF 1: ", frame)
        

class MyUDF_2:
    wish = "ints"
    
    def __init__(self):
        self.local_variable = 23
    
    def process(self, frame):
        print(self.local_variable, "UDF 2: ", frame)
        
uu1 = MyUDF_1()
uu2 = MyUDF_2()

print(uu1.wish)
print(uu2.wish)

for i in make_iterator():
    uu1.process(i)
    uu2.process(i)
```

Some notes on this interface:

 - Initializing the shared-per-task data/buffers in `__init__` doesn't work well, as we then can't instantiate the class without also doing potentially a lot of work. The use-case is instantiating the class on the leader node for allocating the buffers for the results, for which we don't need the per-task data
- Again about the shared-per-task data, they should reside in some kind of namespace, instead of being directly assigned to the instance, otherwise we may get conflicts in the future. Having it in a namespace also means we can "address" the data as whole from the base class or from outside
- I'm currently tempted to release without the wish/constraint system and add support for this later, together with composition. This would implicitly create a default set of constraints (float32, full frames, ...) which can in the future be overridden

sk1p, 2019-05-14:
As an interation on this, how about something like this:

```python
class SomeExampleUDF(UDF):
    # TODO: parameter schema/validation

    def get_result_buffers(self):
        # can access parameters via self.params
        return {
            'super_complex_result': BufferWrapper(
                kind="nav", extra_shape=(self.params.num_components,), dtype="float32"
            ),
        }

    def get_aux_data(self):
        pass
        # TODO: static data that is scattered to all workers. maybe accessible via self.aux?

    def get_task_data(self):
        """
        Initialize buffers/data that is shared for each task.
        """
        return {
            'shared_buf': zeros_aligned(self.params.sig_shape, dtype="float32"),
            'precomputed': None,  # some precomputed value
        }

    def process_frame(self, frame):
        # stupid example, but here it is:
        # self.results would need to have some state, to keep track of the current frame 
        self.results.super_complex_result[0] = np.sum(frame)
        # we can also access `self.params` and `self.data.shared_buf`/`self.data.precomputed`

    def process_tile(self, tile):
        # TODO: still need to figure out the details of this... but it could provide a default impl. that takes care of looping over the tile and calling `process_frame`. 
        pass
```

`process_frame` or `process_tile` will then be called by the libertem internals, what is now in `UDFTask.__call__`.

sk1p, 2019-05-14:
Note that this is still missing the `merge` function, as I'm not yet sure how to best integrate it. It _could_ be as simple as just putting the current merge function as-is as a member of the UDF class... it just has to be made clear that the `merge` method can't access `self.results`, `self.data` etc, as the merge is done on the leader node.

uellue, 2019-05-14:
hmm, what about generally not having results and data as members of the UDF class, but always parameters from outside provided to the function calls?

uellue, 2019-05-14:
> * This would implicitly create a default set of constraints (float32, full frames, ...) which can in the future be overridden

That sounds like a sensible beginner-friendly default. 👍

sk1p, 2019-05-15:
> hmm, what about generally not having results and data as members of the UDF class, but always parameters from outside provided to the function calls?

So basically have only the `params` as attributes of the instance (they are small enough), and always pass in `results`, `aux` and `data`? Sounds like a possibility. Btw. `data` is still too generic as a term, maybe `scratch` is better?

Something like this (also updated for merge function and "validation"):

```python
class SomeExampleUDF(UDF):
    def validate_params(self, params):
        # `params` could be a json-like structure (result of `json.loads`) and this function should
        # validate it and return a cleaned dict, raise some exception otherwise
        # TODO: have a look at existing validation tools
        # do we need access to the dataset here? to check semantic correctness of params?

    def get_result_buffers(self):
        # can access parameters via self.params
        return {
            'super_complex_result': BufferWrapper(
                kind="nav", extra_shape=(self.params.num_components,), dtype="float32"
            ),
        }

    def get_aux_data(self):
        pass
        # TODO: static data that is scattered to all workers.
        # executed on the leader/head node, so we only have access to `self.params`, not `scratch` or `results`

    def get_task_data(self):
        """
        Initialize buffers/data that is shared for each task.
        """
        return {
            'shared_buf': zeros_aligned(self.params.sig_shape, dtype="float32"),
            'precomputed': None,  # some precomputed value
        }

    def process_frame(self, frame, results, scratch, aux):
        # stupid example, but here it is:
        # results would need to have some state, to keep track of the current frame 
        results.super_complex_result[0] = np.sum(frame)
        # we can also access `self.params` and the scratch values we created before, `scratch.shared_buf`/`scratch.precomputed`

    def process_tile(self, tile):
        # TODO: still need to figure out the details of this... but it could provide a default impl. that takes care of looping over the tile and calling `process_frame`. 
        pass

    def merge(self, dest, src, aux):
        # dest is the result for the whole dataset, src is `results` from a single partition
        # TODO: do we need aux?
```

Btw. `aux` and `params` could be unified by just always scattering the `params` (because they are immutable), so we don't have to have these as separate concepts.

sk1p, 2019-05-15:
> Btw. aux and params could be unified by just always scattering the params (because they are immutable), so we don't have to have these as separate concepts.

We also need to consider the case where the user provides one or more values for each frame. Then we should pass the appropriate view as aux value. Maybe like this: if you wrap a `BufferWrapper` around your parameter, the `process_frame` / `process_tile` functions automatically get the right view into that buffer, otherwise, it is passed as-is.

Now the questions would be: 1) should we unify `params` and `aux`? 2) If yes, should `params` also be passed as parameter to all functions (for consistency, as the views will change from frame to frame, like the other arguments)?

uellue, 2019-05-15:
Unifying params and aux sounds good, and the BufferWrapper idea of wrapping aux data to generate views as well.

If I read the code correctly, the UDF class is more of a namespace, then? Why not :-)

sk1p, 2019-05-20:
> If I read the code correctly, the UDF class is more of a namespace, then? Why not :-)

One reason: harder to introduce new parameters. I'll remove the parameter validation/deserialization again, as it is only really useful for integration into the Web API, which is a separate concern. If some basic validation is to be performed, this can be done in `__init__`.

I think together with good documentation this is a feasible approach. I'll follow this up with a first draft PR.

uellue, 2019-05-21:
In order to help with porting our existing UDFs to the new interface, it would be nice to be able to run the "old-style" UDFs in combination with new ones. That will help with debugging and doing the conversion step-by-step.

sk1p, 2019-05-23:
Composability aspects / use cases:

 - correction: dataset-specific processing (example: gain map)
 - pre-processing: analysis-specific processing (example: binning)
 - post-processing/chaining/"child UDFs" (example: working on reduced results, TODO: detailed requirements?)
 - "parallel" execution of independent UDFs (example: calculating a mask application and an average frame in one pass over the data)

sk1p, 2019-05-29:
> In order to help with porting our existing UDFs to the new interface, it would be nice to be able to run the "old-style" UDFs in combination with new ones. That will help with debugging and doing the conversion step-by-step.

Do you think this is still needed? As far as I can see, the only in-tree UDFs that still need to be converted are related to the blob finder; I would actually prefer to have better test coverage there for verifying the functionality.

One backwards-incompatible change is the `init` parameter; in the new UDF we pass a `UDFMeta` instance instead of the `Partition`. That means we can't completely build the old interface using the new one. If you don't use the partition parameter in `init`, you can try something like this (untested):

```python
from libertem.udf import UDF

class CompatUDF(UDF):
    def get_task_data(self, meta):
        if self.params.init is None:
            return {}
        return self.params.init(partition=None)

    def process_frame(self, frame):
        kwargs = {}
        for k in self.params.keys():
            kwargs[k] = self.params._get_view_or_data(k)
        for k in self.task_data.keys():
            kwargs[k] = self.task_data._get_view_or_data(k)
        for k in self.results.keys():
            kwargs[k] = self.results._get_view_or_data(k)
        self.params.fn(frame=frame, **kwargs)

    def merge(self, dest, src):
        if self.params.merge:
            self.params.merge(dest, src)
        else:
            super().merge(dest, src)

    def get_result_buffers(self):
        return self.params.make_buffers()

def run_compat_udf(ctx, dataset, fn, make_buffers, merge=None, init=None):
    udf = CompatUDF(
        fn=fn,
        make_buffers=make_buffers,
        merge=merge,
        init=init
    )
    return ctx.run_udf(dataset=dataset, udf=udf)

```

uellue, 2019-07-11:
A nice feature of composition would be to combine ROIs. Right now, each ROI gets its own pass over the data. That is often inefficient because the ROI doesn't cover all partitions equally, i.e. cores are idle for quite a bit of time. If we bundle UDFs with ROIs, we could get a better balance.

uellue, 2020-04-09:
Since UDFs are now in production for some time already, I'd close this and leave the remaining points as individual issues.

closed 2020-04-09 by uellue