LiberTEM/LiberTEM #287: UDF: Support for auxiliary data

sk1p, 2019-03-07:
Users may want to provide more parameter data, for some example parameters for each frame. LiberTEM should make this data available on cluster nodes efficiently (scatter?), and provide the relevant slice for the current frame to the map function.

sk1p, 2019-03-14:
My current plan:

- rename `ResultBuffer` to `UDFBuffer` (or find a better name: `BufferWrapper` maybe? could maybe be useful outside of UDFs)
- add a method `set_buffer` on `UDFBuffer` to set the buffer to some existing array, which can be used instead of `allocate`.
- similarly to `make_result_buffers`, add an optional parameter `make_aux_buffers` to `map_frames`, which is a function that returns a `dict` of `UDFBuffer` instances. Unlike `make_result_buffers`, the buffers returned here should have data, either via `set_buffer` for existing buffers, or `allocate` and filled, for example if the data is not yet in form of a numpy array
- when calling the frame function, a views for each aux buffers is also passed as argument
- under the hood, the transfer of the aux buffers can be optimized, but doesn't have to be, depending on the executor

uellue, 2020-01-23:
Closing since this is implemented: https://libertem.github.io/LiberTEM/udf/advanced.html#aux-data

closed 2020-01-23 by uellue