LiberTEM/LiberTEM #292: Preprocessing and filtering for Cryo EM

uellue, 2019-03-12:
Discussions with Lothar Houben: Cryo EM generates very large amounts of data. LiberTEM could be a good solution for preprocessing and extracting relevant data from a large-scale raw data set.

As far as I understand, that can involve the following steps:

* Correction step, including gain correction #154 
* Identify relevant portions of a frame
* Alignment of frames
* Extract relevant data, perhaps further reduce, and feed it to reconstruction algorithms

As far as I understood, even being able to generate gain corrected and aligned data on-the-fly from raw data for further downstream processing without generating large intermediate data sets would be a significant advantage.

This issue is just a starting point for the discussion. Implementing this workflow would be a nice project.

sk1p, 2019-03-14:
Do we already have access to example data? Otherwise, maybe we can find suitable data on open data repositories, for example https://www.ebi.ac.uk/pdbe/emdb/empiar/

It would be best if we had a specific and real use-case, maybe even with existing preprocessing results we can compare against.

sk1p, 2019-03-27:
For alignment: https://github.com/bsavitzky/rigidRegistration and related paper