LiberTEM/LiberTEM #293: Hot Pixel Detektion

Algoritter, 2019-03-12:
4D STEM Bilder weisen große Helligkeitsunterschiede auf. Bei den Beugungsreflexen finden sich sehr niedrige Intensitäten (oftmals nur wenige Elektronen-Counts) , hingegen bei dem ungebeugten Strahl sehr hohe Intensitäten. Oftmals ist für die Betrachtung am Bildschirm und für die Datenverarbeitung, wie Edge-Detection, eine logarithmische Skalierung notwendig. Diese Skalierung kann folgendermaßen aussehen:

Log10 (a+b*(Int(x)/IntMax(X)), mit a=1 und b=100

X ist hier ein Frame und x ein Pixel in einem Frame oder auch
X ist eine Matrix und x ein Element der Matrix.

In unserem speziellen Fall findet eine Autokorrelationsfunktion nach der Intensitätsskalierung statt. Das Problem ist, dass wenn ein Hot-Pixel gleich 1 gesetzt wird, alle anderen Werte deutlich nach unten skaliert werden. Insbesondere die Werte der Autokorrelationsfunktion sind dann niedrig.

Siehe Abbildung.
[image]

Wenn mit den gewonnenen Daten eine Karte erstellt wird, wie z.B  eine Darstellung der Spannungen, so enthält die Karte Informationslücken.

Eine Idee zur Lösung des Problems könnte die Bestimmung der Standardabweichung sein. Hot-Pixel haben eine sehr hohe Intensität und hohen Kontrast zu ihrer Umgebung (die Pixel daneben).

Die manuelle Suche von einem Frame unter 256x256 Frames mit einem Hot-Pixel ist nicht einfach. Wenn allerdings eine Stelle auf dem Detektor bekannt ist kann geziehlt in diesem Bereich durch alle Frames gesucht werden.
[image]

Das Ziel wäre es: Die Hot-Pixel zu finden und ihren Wert auf die Helligkeit der Umgebung herab zu setzen damit diese die Auswertung nicht stören. Es kann auch ein neuer Datensatz erstellt werden, der Korrekturdaten für das Datenset bereit hält.

uellue, 2019-03-12:
Hi @Algoritter, thank you for the issue and detailed description! Could you perhaps translate it to English so that other stakeholders can understand it? https://www.deepl.com/translator can help you. 

Many thanks in advance!

Algoritter, 2019-03-12:
4D STEM images have large differences in brightness. The diffraction reflexes have very low intensities (often only a few electron counts), whereas the undiffracted beam has very high intensities. Often a logarithmic scaling is necessary for viewing on screen and for data processing, such as edge detection. This scaling can look like this:

Log10 (a+b*(Int(x)/IntMax(X)), with a=1 and b=100

X is here a frame and x a pixel in a frame or also
X is a matrix and x is an element of the matrix.

In our special case an autocorrelation function takes place after the intensity scaling. The **problem** is that if a hot pixel is set to 1, all other values are scaled down significantly. In particular, the values of the autocorrelation function are then low.

See figure.
https://user-images.githubusercontent.com/48475179/54194850-81480800-44bd-11e9-8931-699da51fb27e.png

If a map is created with the obtained data, e.g. a representation of the stresses, the map contains information gaps. An **idea** to solve the problem could be the determination of the standard deviation. Hot pixels have a very high intensity and high contrast to their surroundings (the pixels next to them).

The manual search of a frame among 256x256 frames with a hot pixel is not easy. However, if a location on the detector is known, this area can be targeted through all frames.

Image with visible hot pixel:
https://user-images.githubusercontent.com/48475179/54195087-14813d80-44be-11e9-9e19-01fd7acc3d4a.png

The **goal** would be to find the hot pixels and reduce their value to the brightness of the environment so that they do not interfere with the evaluation. It is also possible to create a new dataset that contains correction data for the dataset.

sk1p, 2019-03-14:
Thanks for opening the issue!

> An idea to solve the problem could be the determination of the standard deviation. Hot pixels have a very high intensity and high contrast to their surroundings (the pixels next to them).

How exactly do you determine that the contrast is high enough to classify a pixel as hot?

> It is also possible to create a new dataset that contains correction data for the dataset.

See #201 - this would be a use case for materialized derived datasets.

robbmcleod, 2020-01-22:
I discussed methods on how to detect outlier pixels in the original MRCZ preprint:

https://www.biorxiv.org/content/10.1101/116533v1

That discussion was removed from the JSB publication as the reviewers didn't like it and we didn't have the time to argue for it. Basically you look at the histogram and you find where the histogram deviates from the expected Poisson distribution in order to estimate thresholds for hot/dim pixels.

uellue, 2022-07-28:
Ah I see, one could develop an UDF that calculates the histograms. Is this still current?