LiberTEM/LiberTEM #295: Web GUI not working on Internet Explorer

uellue, 2019-03-16:
The GUI doesn't work on Windows 10, Internet Explorer 

[image]

[image]

It does work on Microsoft Edge.

It loads the initial state, but never proceeds to the "start a cluster" dialogue. If the cluster was started with a different browser, the "Browse" button has no effect. 

Ideally, the GUI should work on any ~~modern~~ widely used browser. If support for Internet Explorer is not possible, it should at least display a warning to the user.

sk1p, 2019-03-21:
IE is broken, who would have thought? :grin: Sadly, It's still supported for quite some time, IIRC IE10 support runs out at the beginning of 2020...

In all seriousness, I guess I'm missing some polyfill, or just hitting some edge case... should be fixable in any case.

sk1p, 2019-03-25:
I investigated this a bit more in detail. There were indeed some missing polyfills etc. - unfortunately, this is not all that is required. SVG support in IE11 doesn't appear to be sufficient; focusing an SVG element using javascript [is a pain](https://allyjs.io/tutorials/focusing-in-svg.html#focusing-svg-elements), and only possible using an ugly hack; also, proper sizing of inline SVGs is not possible without some more hacks. Currently, after fixing the missing polyfill issues etc. you get something like this:
[image]

Then, having the wrong size messes up the coordinate calculations. A possible fix is the "padding hack", which unfortunately comes with its own problems.

I fixed the missing polyfill stuff, but I'll postpone further work on this for now; maybe there is a better solution for this I haven't thought about yet. Sadly, feature detecting proper inline SVG sizing could also be quite complex.

As a note; for any old Windows XP machines, I'd propose installing [Firefox 52 ESR](https://ftp.mozilla.org/pub/firefox/releases/52.9.0esr/), if they need to run the LiberTEM GUI.

closed 2023-10-25 by uellue