LiberTEM/LiberTEM #297: Separate developer installation from regular installation

uellue, 2019-03-18:
Discussion with @sk1p: The difference between a developer setup and a regular installation has been a pitfall for novice users. We should move all content related to a developer setup to "Contributing" to keep the two tracks clearly separated.

uellue, 2020-01-23:
@sk1p Is this still a current issue? Perhaps we can re-evaluate and test this with the next round of GSoC when we have to instruct students to set up for development?

sk1p, 2020-01-30:
> Perhaps we can re-evaluate and test this with the next round of GSoC when we have to instruct students to set up for development?

:+1: I think this issue is a good candidate for a first PR from a GSoC student, especially if they had trouble following the docs

Akhileshpm, 2020-02-16:
I would like to contribute.

sk1p, 2020-02-17:
> I would like to contribute.

Hi, welcome to LiberTEM! Note that this issue may have some overlap with #598, so you may need to update your work once that PR is merged.

@Akhileshpm maybe you can get started by laying out a plan how the changes would look like - i.e. what you would move from the general installation docs to the "developer/contributor" installation guide.

Akhileshpm, 2020-02-17:
okay,meanwhile i will contribute in other issue

PR #939 (2021-01-27): Restructure documentation a bit, switch to read-the-docs theme
Closes #297 

Other than switching to the rtd-theme, I've worked a bit on the structure, including splitting the docs into three main sections: user documentation, reference, developer docs. I've also already split the installation to remove installation from git from the normal installation instructions, which already shortens that section by a bit.

## Contributor Checklist:

* [N/A] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [ ] I have  [... cut]
closed 2021-01-27 by sk1p