LiberTEM/LiberTEM #299: Run local cluster tests by default?

sk1p, 2019-03-19:
Currently, tests that spin up a dask distributed cluster are only run if an environment variable is set. Now, I'm adding quite a few of those tests, and I'm not sure if they should still be conditional on the env var. On my laptop, the whole test suite runs in less than 30 seconds, with individual tests taking up to a second. 

@uellue how long do the local cluster tests take on your laptop? Do you think it would be feasible to always run those tests, to gain better coverage in exchange for slower test runs? 

I'd be interested in the output of `pytest tests/server/ --durations=5`.

uellue, 2019-03-20:
@sk1p I'd be in favor of running slow tests by default. The main bottleneck on my machine is getting the environment set up for tox, not the tests. If I am developing, I run selected tests with pytest. It would just be nice to have slow tests together in a few (one?) test files and not scattered throughout the test suite. That way one can run most test files quickly with pytest and doesn't have to pick out or exclude individual tests to avoid running slow ones.

sk1p, 2019-03-20:
> I'd be in favor of running slow tests by default. The main bottleneck on my machine is getting the environment set up for tox, not the tests. If I am developing, I run selected tests with pytest

Okay :+1:

> It would just be nice to have slow tests together in a few (one?) test files and not scattered throughout the test suite. That way one can run most test files quickly with pytest and doesn't have to pick out or exclude individual tests to avoid running slow ones.

For this, I think [custom markers](https://docs.pytest.org/en/latest/example/markers.html) are the way to go. I like having tests of the same functionality close together. 

For some reaaally slow tests, I [already have a `slow` marker](https://github.com/LiberTEM/LiberTEM/search?q=mark.slow&unscoped_q=mark.slow), but for the functional tests that finish in under one (two?) seconds, maybe we need to add a second marker. Then, if you only want to run the fast tests, you can do so via `pytest -m "not functional"`.

uellue, 2019-03-20:
@sk1p sounds good! I'm adding the docs label because I'll probably have to look up how to do that from time to time. Having that in the "contributing" section would be convenient. :-)

sk1p, 2019-03-21:
Done in 50336f9

closed 2019-03-21 by sk1p

uellue, 2019-03-21:
Awesome! :-) 👍