LiberTEM/LiberTEM #305: Use codeclimate.com/quality for LiberTEM?

uellue, 2019-03-22:
At https://github.com/odin-detector/odin-data they use https://codeclimate.com/quality/ 

Among other things, this automatically highlights pasta dishes like this one: https://codeclimate.com/github/odin-detector/odin-data/tools/python/odin_data/frame_receiver_adapter.py/source#issue-78807236f213b2303781407f5616aa04

The service is free for Open Source projects. @sk1p, is this something that could be useful for us?

sk1p, 2019-03-26:
Sounds good. Should we maybe do this after the next release? I already know, it will uncover quite a bit of code duplication in the DataSet implementations :see_no_evil:

uellue, 2019-03-27:
>Should we maybe do this after the next release? 

Sounds good!

> I already know, it will uncover quite a bit of code duplication in the DataSet implementations 

That's the point, I guess! 😆

sk1p, 2020-01-07:
Integration of codeclimate is WIP in #535. For now, the checks run as part of the usual PR workflow. There are at least two more interesting github integrations: [summary PR comments](https://docs.codeclimate.com/docs/github-pull-requests#section-summary-comments) and [automated inline review comments](https://docs.codeclimate.com/docs/github-pull-requests#section-pull-request-review-comments). I suggest we first gain some experience with codeclimate in the current configuration and then decide if these integrations make sense.

Codeclimate supports code coverage summary as well - as we had some flaky experience with our current code coverage tool, it's worth [checking that out, too.](https://docs.codeclimate.com/docs/configuring-test-coverage)

PR #535 (2020-01-07): Add codeclimate configuration
Fixes #305. This only adds a configuration file, specifically to set a tslint configuration file to get rid of [many false positives](https://codeclimate.com/github/LiberTEM/LiberTEM/issues?status%5B%5D=&status%5B%5D=open&status%5B%5D=confirmed&engine_name%5B%5D=tslint) and enable plugins.
closed 2020-01-07 by sk1p