LiberTEM/LiberTEM #306: UDF: detailed documentation

sk1p, 2019-03-24:
Part of #286:

- [ ] move examples from usage to own page
- [ ] explain buffer kinds
- [ ] create a diagram that explains the "mode of operation" - leave out architecture details and work on abstraction level "tasks executed by workers". include data flow and partitioning
- [ ] explain role and parameters of each function in detail, and explain what is gained by this interface compared to a simpler map-like interface without all the buffer handling etc.

jaewshin, 2019-04-09:
I will try to work on this

uellue, 2019-07-11:
@sk1p @jaewshin Perhaps we could review what is still missing from the current documentation to close this issue?

uellue, 2019-09-18:
The current state of UDF documentation should be fine for 0.2, probably? Closing now. We can create new issues for specific docs items when we become aware of them.

closed 2019-09-18 by uellue