LiberTEM/LiberTEM #307: Arbitrary regions of interest in navigation space

sk1p, 2019-03-25:
This is useful for both selections by the user and even more useful for letting analyses work on subsets of the data set. Using a bool array sized like the navigation dimensions could work well internally; we still have to think about the user-facing interface. Possibilities: support rectangular crops, sparse lists of coordinates, ...

uellue, 2019-03-25:
See also the following Issues: 

- Cropping/Culling #28
- Average of frames over ROI #32

sk1p, 2019-03-26:
First prototype is available in [the roi-bitmask branch](https://github.com/LiberTEM/LiberTEM/tree/roi-bitmask), implemented for UDFs (not yet for the mask and sum analysis). ROIs are not yet passed down to DataSets.

sk1p, 2019-03-26:
Rough notes for implementing the I/O side of things:

- as an example: we pass in a mask with 4 bits set to 1 (don't care about which bits,
  right now, let's say random)
- if the tileshape allows, this should result in a single `DataTile` being yielded
  from `Partition.get_tiles`
- so, we have a tile available, but the frames inside may come from very
  different regions of the `DataSet`; their origin no longer conforms to a simple slice. Maybe the slice should even be removed, but I'm not sure how much change this involves
- `BufferWrapper.get_view_for_frame` needs access to the ROI, because it needs to figure out
  where the frames from the tile came from
 - or maybe... `DataTile` should just know this?
 - we need to make sure that `DataTile` always only contains 3D data (or 2D for spectra; important: `shape.nav.dims == 1`)
   otherwise this will break down
 - so, on creation of a `DataTile`, we always pass in a list of the 1D indices into the flattened
   navigation axis, for all frames for which it contains data
 - this should also still make sense for the case where the `DataTile`
   doesn't contain full frames.
 - then, as `BufferWrapper.get_view_for_frame` already gets the `DataTile`, it can easily translate this to navigation coordinates and it should "just work"

uellue, 2019-03-27:
What about just returning a linear list and letting the user figure out how to insert that into a full buffer?

With fancy indexing that is not so difficult: `buf[roi] = raw_data`

Building a "sparse" buffer may not always be what the user wanted. Karina, as an example, wanted to extract a few sample frames for prototyping in #271.

CC #310 :-)

sk1p, 2019-03-28:
> letting the user figure out how to insert that into a full buffer

How about this: we return a `BufferWrapper` or similar object (which we already do, btw!), and accessing `.data` gives the result as a dense array for the whole navigation space; accessing `.raw_data` gives the "linear list"-like result (that is, flattened navigation dimension and shaped like `np.count_nonzero(roi)`). We can make the `.data` property lazy so we don't use any more memory and don't do unnecessary copying.

Or, as we discussed, we use a derived dataset (#201) `ROIDataSet`, which I still have to think through in detail. One thing is certain: with the current dataset implementations, `ROIDataSet` can't be a clean abstraction layer, as the roi needs to be passed down to the dataset to make a difference. I really need to untangle the `libertem.io` mess a bit...

uellue, 2019-03-28:
> How about this: we return a `BufferWrapper` or similar object (which we already do, btw!), and accessing `.data` gives the result as a dense array for the whole navigation space; accessing [...]

Ok, sounds good! :+1:

> Or, as we discussed, we use a derived dataset (#201) ROIDataSet,

I guess we can then disentangle ROI and derived datasets for now, and we stick with your first suggestion. :-)

sk1p, 2019-04-03:
> stick with your first suggestion 

:+1:

Updated notes:

- as an example: we pass in a mask with 4 bits set to 1 (don't care about which bits,
  right now, let's say random)
- let's say the tileshape is also 4, so this should result in a single `DataTile` being yielded
  from `Partition.get_tiles`
- so, we have a tile available, but the frames inside may come from very
  different regions of the `DataSet`; their origin no longer conforms to a simple slice in the dataset or partition coordinates; but it **can** be a slice in the "compressed"/flattened selected ROI coordinates
- UDFs will return a result buffer class, which knows the ROI and can either give the "expanded" results (in dataset coords) or the only "compressed" result buffer, as 1D array (or more precisely: 1 + `extra_shape` D)
 - `DataTile` always only contain 3D data (or 2D for spectra; important: `shape.nav.dims == 1`); actually, all Partitions for all dataset impls should have `nav.dims == 1` (this needs to be enforced somehow, I think, and at least documented. Some `DataSet` impls need to be changed (raw, blo? others?))

Now, moving on to the low level data set:

 - we need to translate from ROI bitmap to indexes, and then read the selected 4 frames into a single `DataTile`. That means frame-by-frame I/O, which is possibly inefficient (need to test!)
- data sets should communicate their capabilities ("can read by ROI"), and there should be a fallback until all impls. support ROIs

More thoughts to follow...

sk1p, 2019-04-03:
There should be appropriate fast-paths for the case of all-1s, this can be determined very quickly with `np.count_nonzero` (or even `np.sum`?)

sk1p, 2019-06-17:
ROIs are available via UDFs, closing. They are, for example, used for "average frame over roi" GUI functionality.

closed 2019-06-17 by sk1p