LiberTEM/LiberTEM #310: UDF: by-tile functions

sk1p, 2019-03-26:
Part of #286.

For some cases, this gives a very nice performance boost over doing things frame-by-frame; as observed by @uellue (and our initial research/prototyping).

Complication: with #307, we cannot easily create a view into the result buffer, as the frames in a tile don't have to be contiguous in navigation dimension any more. So we have multiple options here:

1) instead of `buf[:] = something`, we create a method `buf.set(something)` that fancy-indexes into the result buffer using a list of nav indices
2) implement the slicing interface for `buf[:]` and implement the indirection there. I don't know yet how this should be done in detail; need to experiment. Also, should we support indexing other than the "bulk assign" using `buf[:] = ...`? If the user decides to assign to `buf[i]` for some reason... 
3) don't use a view into the result buffer, make a temporary buffer for the frame result and copy into full result buffer using fancy indexing. may have significant overhead if you have a) many result buffers or b) larger results per navigation position

For consistency, I'm currently leaning towards either 2), or 1) with corresponding change to the by-frame UDFs.

If this is complete, we may want to port over our current analyses to this interface. This of course also requires integration into the Web API.

sk1p, 2019-03-26:
> If the user decides to assign to `buf[i]` for some reason...

I think this is actually possible to solve: we index into the coodinate list, and then use the resulting slice of coordinates as fancy index to do the actual assignment operation. We "just" need to make sure this works generally; operations other than assignment need to work, too.

sk1p, 2019-03-29:
I just had an idea for an alternative interface: why limit ourselves to "by tile", why not provide a generic "iterator of frames" interface? This may allow to express some problems more clearly, and keep state in local variables without doing buffer management. Basically, this would move parts of the innermost loop of `UDFTask` into the udf itself.

The object returned by the iterator could include a view into the result buffers as well (in this case, that object is just a tuple, but it doesn't have to be):

```
some_variable = 42
for frame, results in frame_iterator:
    some_variable += frame.reshape((-1,))[random.randint(0, frame.size)]
    results['some_navbuffer'][:] = some_variable
```
(excuse the stupid example)

This is just thinking out loud... are we losing any optimization opportunities by not providing contiguous tiles at all? Yes, because then we can't call `numpy` operations on whole tiles. So there is value in a by-tile interface, too.

This proposed interface also gets rid of the `init` function, because you can just do initialization before the for-loop (`some_variable = 42`).

uellue, 2019-03-30:
Perhaps we can write mock-up code for the same application in both ways and see which one is shorter? Right now I can't see much of a difference between both ways.

An advantage could be that local variables can be managed more efficiently and easily, right?

I do think that per-tile functions are still a very promising concept because they would allow us to perhaps even be faster than plain numpy with their cache efficiency.

uellue, 2019-04-02:
@sk1p and I came up with almost the same code simultaneously. :grin:

This code example uses a coroutine, which is one of the more obscure Python features: A generator, just in the other direction, i.e. consuming data.

Advantages:
* Initialization and local variables are natural and simple
* We can process a data block with several UDFs in a row for optimal cache efficiency, enabling efficient composing
* To keep things compact, We can communicate any requirements or capabilities regarding the input data -- full frames, frame stacks, tiles, full partitions, ...
* To keep things compact, we may even be able to communicate requirements of our output buffer

Disadvantages:
* Using an obscure feature, requires deeper understanding of Python 
* Communicating the requirements like this looks quirky to me

Alternatives:
* Classes?

```python
# The coroutine can communicate requirements to the "host"
class Wishlist:
    def __init__(self, mywish=None):
        self.mywish = mywish
        
    def wish(self, mywish):
        self.mywish = mywish

def udf_1(wishlist):
    wishlist.wish("floats")
    while True:
        frame = (yield)
        print("UDF 1: ", frame)

        
def udf_2(wishlist):
    wishlist.wish("ints")
    while True:
        frame = (yield)
        print("UDF 2: ", frame)

def make_iterator():
    yield "Frame 1"
    yield "Frame 2"
    yield "Frame 3"
    yield "Frame 4"    

wishlist_1 = Wishlist()
wishlist_2 = Wishlist()

u1 = udf_1(wishlist_1)
u2 = udf_2(wishlist_2)  

# Prime the coroutines
next(u1)
next(u2)

print(wishlist_1.mywish)
print(wishlist_2.mywish)

for i in make_iterator():
    u1.send(i)
    u2.send(i)
    
u1.close()
u2.close()
```

Output:

```
floats
ints
UDF 1:  Frame 1
UDF 2:  Frame 1
UDF 1:  Frame 2
UDF 2:  Frame 2
UDF 1:  Frame 3
UDF 2:  Frame 3
UDF 1:  Frame 4
UDF 2:  Frame 4
```

uellue, 2019-04-02:
Implementation with object orientation:

```python
class MyUDF_1:
    def __init__(self):
        self.local_variable = 42
        
    def wish(self):
        return "floats"
    
    def process(self, frame):
        print(self.local_variable, "UDF 1: ", frame)
        

class MyUDF_2:
    def __init__(self):
        self.local_variable = 23
        
    def wish(self):
        return "ints"
    
    def process(self, frame):
        print(self.local_variable, "UDF 2: ", frame)
        
uu1 = MyUDF_1()
uu2 = MyUDF_2()

print(uu1.wish())
print(uu2.wish())

for i in make_iterator():
    uu1.process(i)
    uu2.process(i)
```

Output:
```
floats
ints
42 UDF 1:  Frame 1
23 UDF 2:  Frame 1
42 UDF 1:  Frame 2
23 UDF 2:  Frame 2
42 UDF 1:  Frame 3
23 UDF 2:  Frame 3
42 UDF 1:  Frame 4
23 UDF 2:  Frame 4
```

uellue, 2019-04-02:
User testing: @woozey found the class-based approach more readable and understandable at first glance.

sk1p, 2019-05-22:
Ok, so the discussion above was a bit derailed :grin: and now is quite outdated.

Anyways, I have by-tile processing now implemented, and also set `full_frames=False` for this case, which means at least the K2IS data is properly tiled. There, the performance difference is quite noticeable; a tiled pixelsum (summing up all pixels in a frame) over a 8GB data set, on my laptop takes 2.48 s, where the by-frame method takes 4.39 s. I think this can be attributed mostly to cache effects: when reading whole frames, they can't be held in L3 cache in decoded form, so they spill over to the main memory.

Another question remains, though: is the current interface good enough for tiled operations? For example, the `stddev` UDF by @jaewshin uses a `kind='single'` buffer for recording how many frames it has seen. For `full_frames=False` tiles (tiles that are not stacks of full frames, but stacks of cropped regions), we would need independent counters for all possible crop positions. With "some" memory waste, it can already be implemented with a `kind='sig'` buffer, but that is of course not optimal. Maybe the dataset should give some information about the structure of tiles (example: have a method on the partition that answers "if you ask me for tiles of this dtype in the future, there will be X possible positions", and then provide a simple index on each tile.)

Thoughts?

sk1p, 2020-01-21:
> Maybe the dataset should give some information about the structure of tiles (example: have a method on the partition that answers "if you ask me for tiles of this dtype in the future, there will be X possible positions", and then provide a simple index on each tile.)

See #554 for a proposed interface that could include tiling information. Still TODO: draft for a tiling scheme "encoding".

uellue, 2020-01-23:
Closing here since we have a more specific discussion in #554

closed 2020-01-23 by uellue