LiberTEM/LiberTEM #325: distortion correction via matrix multiplication

FWin22, 2019-04-18:
It would be great if center-of-mass and other functions allow for a proper scaling of the k-vectors in the diffraction pattern.
Ideally, the functions would accept a transformation matrix as an argument, which is then multiplied to the k-vectors of the respective diffraction pattern:
`M = np.array([a, b], [c, d]); k_new = np.dot(M, np.array((kx, ky))`
This would allow to correct for arbitrary rotations and distortions, as well as proper scaling of the sampling rate, i.e. converting from pixels to reciprocal nm.

The implemantation should be fairly straightforward i guess.
What do you think? @uellue @sk1p

sk1p, 2019-04-23:
Thanks for the suggestion! As discussed in person, for virtual detector based analyses, the transformation matrix could be applied to the mask, as we don't want to transform/resample the whole dataset. The interface provided should be uniform, i.e. all analyses should be able to access the correction matrix.

For center of mass, we can transform the result vector field, but we also have to take into account the transformation of the radial cut-off mask.

I think this could be the start of our metadata/units support (#121), and the correction matrix could be set as metadata to the `DataSet` (in the future: read from the metadata in the source files, if possible).

How should the integration into the GUI look like? Should we have a workflow for interactively determining the correction? Doing this for rotation should be easy enough, if not very precise; shear can be quite subtle...

Related: #31

FWin22, 2019-04-23:
I don't have a good idea on how to implement this for the GUI.
However, as in the API, it should also be possible to just input the matrix manually, i.e., the values of each matrix element.
The matrix for distortion correction (without rotation) will mainly be measured individually for each microscope/camera length and can then be regarded as constant. So i guess most people, who will use this feature, will know their matrix in advance.
Regarding rotation and scaling, one could setu the matrix by the rotation angle and sampling density (nm^-1 per pixel).