LiberTEM/LiberTEM #328: UI: saving of the analyses

woozey, 2019-04-29:
Although it has been discussed several times, just want to see some feedback and that this topic stays somewhere in the TODOs:

As UI is very handy for interactive analyses, would be great to have several saving possibilities:

* Save raw results of the analyses ideally in several formats as an option: npy, hspy, raw binary, libertem object, ... - ?;
* Save jobs that one can run them using api;
* Save job parameters in some other way: text, python dictionary (npy-?).

sk1p, 2019-04-29:
Thanks for bringing this up again. Yeah, we definitely want to have this feature (See also #46). Maybe I can squeeze this in between some of the larger issues. Required for most of this is actually keeping the results on the server, tracked in issue #168.

Exporting in most formats should be easy enough, it is mostly the changes from #168 this is waiting on. All three options sound sensible. Job parameters could be exported as JSON, which is a natural fit here.

> Save jobs that one can run them using api

Export for running via API is a good idea (#158), should have the option to include all required boilerplate code (creating context, loading dataset, ...) and provide the code in a copy-pastable text area. This functionality could be provided by the `Analysis` classes. This "python script export" can be done independently of #168, mostly some GUI work. Maybe this would be a good first step?

woozey, 2019-04-29:
Great! I knew that I'm not the only one, but didn't expect full redundancy in this issue))) Have been using keyword "save" in the issues search whereas "export" would have brought me to better results...

bdalevin, 2020-02-26:
I would also like to comment that it would be extremely useful to be able to save raw results of the analyses from the GUI. 
float .tif images would be very useful in addition to the formats already mentioned above. 
Please implement this as soon as you can.

closed 2020-03-18 by sk1p