LiberTEM/LiberTEM #335: Is there an efficient way to share data between workers?

uellue, 2019-05-14:
MaskJobs that match a dataset against a large template library may benefit from sharing the library between workers on a node. In particular, L3 cache use on machines with many cores should be much improved. Is there an efficient and portable way to achieve this?

I was reading through the dask.distributed documentation, and it seems to be possible to share data with persist() and scatter(). However, I couldn't figure out if the data is copied to the workers or resides in shared memory.

What would be the best way to test the behavior of dask.distributed? Or is there a more efficient and easy way to share the data?

sk1p, 2019-05-14:
> Is there an efficient and portable way to achieve this?

Hmm, something like [this](https://docs.python.org/2/library/multiprocessing.html#multiprocessing.Array)? 

> I was reading through the dask.distributed documentation, and it seems to be possible to share data with persist() and scatter(). However, I couldn't figure out if the data is copied to the workers or resides in shared memory.

AFAIK, the data is copied (I could be wrong, though! need to check the source), but using `scatter()` should already a first step into the right direction, as currently static auxiliary data (like masks) is sent to the worker process with each task (see also #287).

Did you try to run dask.distributed workers with threads instead of processes? Then, if we scatter the data, it should be shared, or at least possible to share.

uellue, 2019-05-14:
> Hmm, something like [this](https://docs.python.org/2/library/multiprocessing.html#multiprocessing.Array)?

Yes, that could be an option. One would have to coordinate the workers on a node so that only one creates the shared memory and the others receive handles to it. I was hoping that someone else already implemented that...

> AFAIK, the data is copied (I could be wrong, though! need to check the source), but using `scatter()` should already a first step into the right direction, as currently static auxiliary data (like masks) is sent to the worker process with each task (see also #287).

Yes, I was also assuming copying, but don't know for sure. I haven't tried it yet, I wanted to ask before doing work.

uellue, 2019-05-15:
Another option could be something like a dataset. That would be a rather natural way to make larger persistent data available at workers.

sk1p, 2019-05-18:
Just as a note, the [ray distributed AI framework](https://ray.readthedocs.io/en/latest/index.html) uses a [shared-memory object store](https://ray.readthedocs.io/en/latest/plasma-object-store.html) for this purpose.

uellue, 2019-06-06:
As a remark, this is becoming relevant for the Radial Fourier Analysis #356 . Mask stacks for K2 frames are becoming so large that building them on 18 cores in parallel may take longer than applying them, depending on the scan size.

sk1p, 2020-01-20:
What would need to be done to implement this, using the plasma object store (adding a dependency on pyarrow):

 - [ ] Add a new script entry point `libertem-worker` for starting libertem workers, which is basically a wrapper around `dask-worker`, which additionally starts an instance of the plasma object store
 - [ ] `libertem-server` also needs to start the plasma object store, but only when using the local cluster
 - [ ] add simplified low-level functions for `read`/`write` access to the object store
 - [ ] add new method(s) to UDF interface for specifying/generating static data
 - [ ] generate and store the static data on all nodes before running a UDF (via `run_each_host` or similar), and clean up afterwards
 - [ ] `UDF.static` is then a proxy-object, which only knows about the shared data by id, and executes the `read` function for said id on access

For sparse arrays / matrices, we may need to store more than one id, and some metadata.  I guess this comes down to having some kind of `serialize_sparse` and `deserialize_sparse` functions, which don't touch the actual data, just serialize down the structure (what kind of sparse array (csr, csc, coo, ...), which impl, ...). Needs some specification work here.

uellue, 2021-02-24:
@sk1p is this the discussion about shared memory in Dask that you mentioned? https://github.com/dask/distributed/issues/4497

sk1p, 2021-02-24:
Yup, that's it, thanks for linking it here.

uellue, 2024-05-16:
Using https://distributed.dask.org/en/stable/api.html?highlight=scatter#distributed.Client.scatter has pretty much solved this issue IMO, closing

closed 2024-05-16 by uellue