LiberTEM/LiberTEM #338: Generate a deep mask stack with one function instead of list of functions

uellue, 2019-05-15:
Currently, mask stacks are assembled one by one. This works well for scenarios with few masks, such as shape or COM analysis.

For other purposes, such as radial integration, non-trivial binning or template matching, a deep stack of masks with thousands or tens of thousand masks could be useful. Tests for #197 have shown that this can be efficient if the matrices are sparse.

For that reason, we should accept functions that generate the whole stack at once instead of slices.

Possibilities for implementation:
1. Accept both list of functions or single function for factories argument. Advantage: minimal impact. Disadvantage: Magic polymorphism, not obvious
2. Two kwargs, factories or factory, of which one is set. Advantage: Explicitly clear which is what. Disadvantage: Two invalid situations -- bot set or both not set
3. New subclass of mask analysis. Advantage: Clear and explicit. Disadvantage: Significant code duplication and growing API surface.

I am tending towards the first one. What do you think?

sk1p, 2019-05-15:
I would also prefer the first one. So the idea would be to either pass a list of callables, or a callable that returns a list? Or should the single callable return a ndarray (which can be sparse or dense?) instead of a list?

I think with that interface we cover quite a lot of ground already. How does this integrate with the `MaskContainer`? I guess it just needs some changes how the `raw_masks` are computed?

> Tests for #197 have shown that this can be efficient if the matrices are sparse.

So the problem there is actually calling the many functions to "materialize" the masks? Or is it also the serialization of all the functions?

uellue, 2019-05-15:
The single callable should return a single ndarray resp. sparse matrix.

Yes, as far as I can see only the mask computation, slicing and assembly.

Actually, I didn't test performance. My main motivation is to not artificially generate thousands of functions, but to just write one function that does the job and keep the code simple and clean. Performance should be a topic, too, though. Preferentially I'd like to be able to use numba since that should be easier to understand compared to numpy if the mask calculation becomes complex.

sk1p, 2019-05-15:
> My main motivation is to not artificially generate thousands of functions, but to just write one function that does the job and keep the code simple and clean

:+1:

> Preferentially I'd like to be able to use numba since that should be easier to understand compared to numpy if the mask calculation becomes complex

Sounds good.

As for performance, I also thought about doing some re-ordering of the masks to improve data locality (similar aim as [this paper](http://citeseerx.ist.psu.edu/viewdoc/download?doi=10.1.1.308.6067&rep=rep1&type=pdf), I guess?). Basically, if two masks in the stack access the same data, they should be close to each other in the stack. That would be a topic for future optimization, maybe also directly in `sparse`.

PR #339 (2019-05-20): Allow creating a stack of masks with one factory function
Still to do:
- [x] Update perforamnce test notebook from sparse
- [x] Update or implement helper functions in libertem.masks

Closes #338
closed 2019-05-20 by sk1p