LiberTEM/LiberTEM #357: File read dimensions problem

matkraj, 2019-06-07:
I have a file which contains 256x256 + 2 diffraction patterns due to additional triggers from scanning system.
I would like to be able to read this in LiberTEM.
In the current git version, this fails with a complain that the number of images in the file does not match the geometry. This should be only an upper limit -> e.g. we would be able to even read first 20 lines or so if needed.

matkraj, 2019-06-07:
can be fixed by changing:
https://github.com/LiberTEM/LiberTEM/blob/15d190dbeaa6b4baea762d90515bdf60a7e41f3d/src/libertem/io/dataset/mib.py#L352
to 
`if s[0] * s[1] > num_images: `

uellue, 2019-06-08:
@matkraj thanks for reporting and suggesting a fix! Just to confirm, you tested it with your file and it resolved your issue? In that case, would you like to file a pull request so that you are credited as contributor? I believe that even works by editing the file on GitHub.

sk1p, 2019-06-11:
Thanks for filing the issue! This change should also have a unit test, to prevent regressions in the future. It could try to open the test data set with `scan_size=(16, 32)`, which fails right now in `master`.

Currently, we don't run the I/O tests in continuous integration because we don't have a good way of distributing the test data, and often we don't have clearly licensed data sets available (see also https://github.com/LiberTEM/LiberTEM/issues/86 - help in this area is appreciated!). If you want to run them locally, you need to make a `(32, 32)` MIB file available as `<your_libertem_clone>/data/default.mib`. The file I'm currently using contains a test pattern ("Merlin diamond" as text) and you may know it - I got it from @enebotde

Let me know if you need help, with tests or [pull request creation](https://help.github.com/en/articles/creating-a-pull-request-from-a-fork)!

matkraj, 2019-06-12:
I am also not 100% sure it should be a fail if the data is shorter than the geometry. I'd suggest a warning and the data should be analysed in any case (grey signal for missing points).This could happen if something went wrong during the acquisition or we try to analyse unfinished data.

sk1p, 2019-06-12:
That's a good point. So it's also about properly handling missing data (for which we have https://github.com/LiberTEM/LiberTEM/issues/256), specifically when reading MIB files.

Do we only need to expect data loss (or too much data) at the end of the dataset, or is there any plausible scenario where e.g. a .mib file from the middle of a dataset can go missing? I know we can recover from that also (check if the sequence number of file N matches number of images plus sequence number of N-1), but I wonder if it is likely to happen.

closed 2019-06-18 by sk1p

sk1p, 2019-06-18:
I incorporated the fix for the initial issue; having less data than the `scan_size` will then be fixed as part of #256

commit db8f2169 (2019-06-18): MIB: Fix for mismatch between scan_size and number of images

Fixes #357. In the future, we also want to support MIB files with less
frames than needed for the scan_size, I consider that part of #256

Thanks @matkraj for the report and fix!