LiberTEM/LiberTEM #363: Support for dimension reduction

jaewshin, 2019-06-14:
Dimension reduction is a useful technique for understanding patterns in the data. If implemented, dimension reduction methods will be integrated into UDF interface. To be integrated into UDF interface, dimension reduction algorithm must satisfy two conditions: (1) can fetch a frame or a collection of frames in an iterative manner, and (2) can merge independent outputs of the algorithm performed on each partition. More specifically, the point (1) refers to an algorithm that can build results on top of previous results with addition of new (set of) frame(s). While the point (2) may sound similar to the point (1), these two points are not equivalent since an algorithm that builds results in iterative manner is not the same as an algorithm that merges outputs computed from multiple disjoint batches of data. Lastly, we must take into account the memory space issue. For instance, the most common way of computing Principal Component Analysis is to use co-variance matrix of the data, but if the number of pixels of a frame is 256 x 256 = 65536, we need 2147483648 entries to store such a matrix, even after reducing for the symmetric property of the matrix. Thus, an algorithm must not use too much memory space for every step of its computation.

Currently, possible candidates for dimension reduction methods are Principal Components Analysis (PCA) and Non-Negative Matrix Factorization (NNMF), and various considerations for each method is described below:

1. Principal Component Analysis (PCA)
The current plan of action is to separate the two stages of UDF (into processing frames and merging partitions) and apply different algorithms for different stages. 

As for the first stage of processing frames, many iterative version of PCA is available. Among others, Incremental PCA (Incremental Learning for Robust Visual Tracking, https://www.cs.toronto.edu/~dross/ivt/RossLimLinYang_ijcv.pdf) and Candid Covariance-Free Incremental PCA (http://citeseerx.ist.psu.edu/viewdoc/download?doi=10.1.1.7.5665&rep=rep1&type=pdf)  seems to be the most suitable options. Both algorithms are similar to one another, but the latter appears more promising since it is 'free' of covariance matrix and thus can avoid the memory issue more easily. Note however that we still have the issue of managing the total number of frames. A dataset may contain 20K ~ 65K number of frames which, with the current methods, cannot be further reduced and can pose challenge to memory.

Unfortunately, there does not seem to be an efficient method to combine two or more PCA results. So far, I was able to find two algorithms Distributed PCA and k-means clustering (http://www.cs.cmu.edu/~ninamf/papers/distributedPCAandCoresets.pdf) and merging of SVD results that is used in gensim package for topic modeling (https://radimrehurek.com/phd_rehurek.pdf - pg 45). The first one, distributed PCA, is actually an algorithm that performs 'local PCA' for each partition and then merges the 'local PCA' results through 'global PCA'. The 'local PCA' cannot be directly placed under UDF interface because it assumes access to full-batch of the partition. 'Global PCA' is a sound option for merging PCA results with known error bound, but it makes use of covariance matrix so memory becomes an issue again. The second algorithm for merging SVD results can be directly implemented in the UDF interface. However, we need to carefully check if the method is mathematically sound. 

2. Non-Negative Matrix Factorization (NNMF)
NNMF has several parallel algorithms available. However, most parallel algorithms use partitioning of the columns of the data matrix (https://arxiv.org/pdf/1509.09313.pdf). Under LiberTEM setting, we can view a data matrix as follows: each row represents a 'flattened' frame where each column in the row corresponds to pixel values. Then partitioning the columns of the data matrix is equivalent to partitioning the pixels of a frame, which is not really feasible with the current architecture. This paper (http://www.ambuehler.ethz.ch/CDstore/www2010/www/p681.pdf) appears to have a more promising outlook in that it partitions along whichever axis has the smaller dimension (e.g., if the number of frames is less than the number of pixels in a frame, it partitions along each individual frames). 

After implementation, we also need a sound way of testing each functions. Note that most of the methods mentioned above are approximation methods to the true estimation and some even has randomness embedded in their algorithms. One way is to perform full-batch PCA or NNMF on a test dataset and check whether an 'error' is below certain threshold. There can be several candidates for the metric of evaluating such an 'error', with frobenius norm being the most commonly used option. However, it remains for us to find out what threshold can be considered sufficient, as it can heavily depend on the size of the data, method we are using, etc.

jaewshin, 2019-06-14:
See #347 for the current implementation of PCA. It uses Candid Covariance-Free Incremental PCA for process_frame and merging SVD, as suggested by https://radimrehurek.com/phd_rehurek.pdf, for merge (i.e., merging partitions).

sk1p, 2019-06-17:
> See #347 for the current implementation of PCA. It uses Candid Covariance-Free Incremental PCA for process_frame and merging SVD, as suggested by https://radimrehurek.com/phd_rehurek.pdf, for merge (i.e., merging partitions).

Nice! Using a covariance-free version sounds like a sensible plan. What is the trade-off of using incremental PCA locally versus having "access to full-batch of the partition"?

sk1p, 2019-06-17:
Regarding NMF:

> This paper (http://www.ambuehler.ethz.ch/CDstore/www2010/www/p681.pdf) appears to have a more promising outlook in that it partitions along whichever axis has the smaller dimension (e.g., if the [...]

So, one could modify this algorithm to always partition in a way that is compatible with LiberTEM's partitioning?

jaewshin, 2019-06-18:
> What is the trade-off of using incremental PCA locally versus having "access to full-batch of the partition"?

I'm assuming you are talking about using incremental PCA for each observation (i.e., frame) versus using it for a set of observations. Incremental PCA was designed for memory-efficiency purpose so that one doesn't have to use too much resource when doing large matrix operations. I guess it depends on how big of a full-batch for a partition is. If it is of moderate size, then it could boost the performance.

jaewshin, 2019-06-18:
> So, one could modify this algorithm to always partition in a way that is compatible with LiberTEM's partitioning?

I actually don't think possible to use the algorithms in the paper... I'm still trying to understand it, but it appears that you need some way of accessing a result from a worker (i.e., result from operations on a partition) from another worker when processing partition.

uellue, 2022-07-28:
Closing for now, it seems that there are few activities in this direction at the moment. Also, reducing the data before doing PCA/NMF/... rather than performing it on the whole raw data seems to work better.

closed 2022-07-28 by uellue