LiberTEM/LiberTEM #370: TimeoutError in CI resolved by nannynanny()?

uellue, 2019-06-18:
The CI TimeoutError phenomenon hasn't occurred since we merged the PR #360 with nannynanny() waking up the loop regularly. Should we perhaps undo all the other changes, namely pinning dask.distributed to a lower version:
https://github.com/LiberTEM/LiberTEM/blob/8a93f7bc451637506a25b2466ad5ae51a08d5abb/setup.py#L136

and setting the thread count to one
https://github.com/LiberTEM/LiberTEM/blob/8a93f7bc451637506a25b2466ad5ae51a08d5abb/src/libertem/executor/base.py#L120

and see where that takes us?

sk1p, 2019-06-18:
I would prefer to do that in 0.3 (or 0.2.1 or whatever we decide to release next)

sk1p, 2019-12-12:
CI has been running smoothly for the last time, with exception of the time mocking anomaly, which should now be fixed.

closed 2019-12-12 by sk1p

uellue, 2019-12-12:
What about the ThreadPoolExecutor? I'll do a PR to change it back, to see what that does.