LiberTEM/LiberTEM #380: Improve flow control for websocket connection

sk1p, 2019-06-25:
The flow control was already improved recently (#369, #367), but it is not completely done yet. One reason we still send too much data in case of a bad connection: `broadcast_event` is kinda-sorta asynchronous (the underlying methods don't *require* `await`, but they return immediately and send the data asynchronously).

To actually get back pressure from the websocket, we need to gather the futures returned by `write_message` in `broadcast_event`. This is now actually done as part of https://github.com/LiberTEM/LiberTEM/commit/17481afaa0a1d13187ed2461cb4fe87319e6f1f7 - the `send_results` method only returns once the results have actually been sent (meaning: copied to the socket buffers of the operating system).

I could not test this yet - @uellue: maybe you can do the same tests you did before with the new version and check if the behavior improved?

Related to #134

uellue, 2019-06-28:
My test is very crude, so one should be careful with interpretation. I have a feeling that it does improve somewhat and that the update rate is lowered rather aggressively, as it should. However, there are still several GUI updates waiting in some queue. In the terminal I can see that a job is long completed, but I still get several pending GUI updates.

sk1p, 2019-07-01:
There are also the built-in tools in browsers for network throttling, but sadly both [firefox](https://bugzilla.mozilla.org/show_bug.cgi?id=1421357) and [chrome](https://bugs.chromium.org/p/chromium/issues/detail?id=423246) don't support throttling websockets

uellue, 2022-07-28:
@sk1p is this still current?

sk1p, 2022-07-28:
Probably - I don't think much changed in that area. But at least chrome did get websocket throttling support this year, so it should be easier to test.