LiberTEM/LiberTEM #381: Improve performance in GUI

sk1p, 2019-06-26:
Test done: apply simple mask (all 1s) from API, repeat same from GUI (with disk mask analysis, maximum radius). All from filesystem cache.

Data: EMPAD 4GiB, MIB 4GiB, FRMS6 8.6GiB

Observation: the Python API is fast-ish (~0.2s for EMPAD, ~1.6s for MIB, ~8.1s for FRMS6), while the GUI is sometimes drastically slower (~0.5s for EMPAD, ~9s for MIB, ~30s FRMS6).

So, two things: 1) something makes processing in the GUI slower, dependent on the file type, 2) the FRMS6 reader is not yet as fast as it could be (IMHO, even though it does a bit more work, like dark frame substraction and unfolding the weird half-frames)

Can't say I have found the culprit yet, but one more observation: the (UDF-based) average frame calculation is also affected, and there disabling visualization of intermediate results helps, but the same is not true for the mask job result, even throwing away all results and not reducing them into the result buffer doesn't help.

Possible next steps:

 * need to do some system-wide perf profiling again
 * write a mask UDF and compare perf. to apply mask job 
 * run libertem server on moellenstedt/madmax and GUI on laptop and see if perf. is better
 * implement some basic performance tracing (for example, emit the [chrome event tracing format](https://docs.google.com/document/d/1CvAClvFfyA5R-PhYUmn5OOQtYMH4h6I0nSsKchNAySU/preview) which can be visualized in chrome://tracing/)

sk1p, 2019-06-27:
> one more observation: the (UDF-based) average frame calculation is also affected, and there disabling visualization of intermediate results helps, but the same is not true for the mask job result, even throwing away all results and not reducing them into the result buffer doesn't help.

This is actually because the mask job result has a busy indicator on the apply button, whereas the average frame visualization only has the spinner on top of the result image. On Chrome, this has a huge performance impact somehow (maybe related to my system configuration, too?), which I have yet to properly debug/understand.

This is also visible in the system wide profile, where the slow case for the average frame calculation had chrome and related programs visible. The fast case was dominated by the Python processes of LiberTEM. Taken using `perf record -a -g -- sleep <something>`, processed by `perf script` and visualized using [flamescope](https://github.com/Netflix/flamescope) and selecting the "hot" parts:

Slow:
[image]

Fast:
[image]

Using Firefox, the performance impact is still there, but much less severe (at least a factor 2 less impact than chrome).

Disabling all busy indicators and updates for partial results confirmed this: both Firefox and Chrome are almost exactly as fast as the scripted version. Here is a profile from Chrome (with intermediate results disabled, so this is about the best case):

[image]

As you can see, as indicated by the green bars in the "Frames" row, there are only redraws at the beginning (as I was clicking the apply button and moving the mouse out of the window); afterwards, only when the full results are received, a redraw occurs.

Compare to the case with running animations:

[image]

The Frames row is completely green, as we need to render 60 times per second. Zooming in a bit:

[image]

We can see that most frames even take more than 16ms, so the 60fps target isn't even hit. I'm going to stop this session here and next try to find out why Chrome is so much slower than Firefox, and maybe also look at it on different systems.

Btw: look at [these beautiful flamescope patterns!](http://www.brendangregg.com/blog/2018-11-08/flamescope-pattern-recognition.html)

uellue, 2019-06-28:
This is very, very interesting. 👍👍👍

uellue, 2022-07-28:
@sk1p is this still current?

sk1p, 2022-07-28:
When I last tested this, it was on a different laptop - on my current one, I can't reproduce a significant performance difference between firefox and chrome - I'll close it for now. Some food for thought - in some situations performance may improve by having _less_ worker processes, and having a bit leftover compute for browsers etc.

closed 2022-07-28 by sk1p