LiberTEM/LiberTEM #382: Adaptive partition sizes

sk1p, 2019-06-27:
For more computationally intensive workloads, it would make sense to show results for smaller units. For this, an adaptive approach could work: start with small partition size and scale it up until we arrive at a nice interactive rate. Could also adapt to load changes on the computation nodes.

For this to work, we need to separate partitions as unit of work and partitions as unit of storage for data locality.

sk1p, 2020-02-20:
Design: could have an iterator class that is returned by `DataSet.get_partitions` which has as state the current partition size "wish". The tasks need to be submitted in batches, and based on the runtime of a batch, the partition size should be adjusted. This affects the size of partitions that are generated from the iterator afterwards.

Should first implement the interface and general functionality and slowly migrate datasets to actually adhere to these wishes.

sk1p, 2021-11-11:
I've edited the issue to include a first stab at a spec for these changes - if this sounds sensible, I'll implement a prototype.

/cc @uellue  @matbryan52 anything to be added from your perspective? Any pain points regarding our `dask` integrations?

uellue, 2021-11-12:
The summary sounds good!

> The performance trend needs to keep in mind differences between GPU and CPU.

That's an interesting one since we can even have different performance per worker on the same device class. Probably it will be enough to track average performance per class, though, since we don't have to be 100 % precise with the update rate, just in the right ballpark.

> How does this interact with live processing?

In my gut feeling it will play together quite nicely. I just don't see it 100 % yet how it should be done. With a distributed data source we'd have to make sure that all source nodes agree on the size of a specific partition. I'd leave this as a next step since adaptive partition size will probably be useful for offline data as well and doesn't necessarily have to be used for every source, so we can figure this out later.

I was also thinking about how to determine the starting partition size. If we start too large, we would "stare at an empty result in the GUI", and if we start too small we reduce max performance through overheads and congestion at the main node. What do you think about starting with a range of partition sizes from a min to a max size? That would have two benefits:

1. We'd make sure that we get at least a few updates quickly and can adapt the size for the following steps
2. The results don't all arrive at the same time, meaning the main node can merge and send out new tasks without causing workers to wait.

For small datasets it would probably be smart to reduce partition size in powers of two so that we can always stack them without overlap, assuming that twice the partition size takes roughly twice the time. Otherwise we might have stragglers.

If that is not practical, for example if we only have a single eligible worker or are doing live processing, we should probably start relatively small, right?

We should also track how much time updates take on the main node, similar to the rate limiting for the GUI. If the main node can't keep up, we can reduce the load by working on larger partitions. The main process should be waiting for new results at least for a short time between updates.

uellue, 2021-11-12:
> The main process should be waiting for new results at least for a short time between updates.

Probably we should have a moving average since individual updates arrive stochastically, or even in bursts if we send out many tasks of the same size at the same time.

uellue, 2021-11-12:
> Any pain points regarding our `dask` integrations?

For the `DelayedJobExecutor` and `make_dask_array()` we'd still use constant partition sizes.

A `DaskDataSet` is quite restricted in its partitioning, as far as I could see. Technically it could ask Dask to give it one large chunk as a "raw partition" and then slice smaller partitions from it? However, that would mean with the current model that we don't have "1 Task = 1 partition" anymore, since we'd have to return several partition results from the same intermediate NumPy array that Dask gave us.

As an alternative, one could fire off the reading process for a "raw partition" as a separate Future, which is then used as an argument to pass the partition data as an argument into several tasks. The Dask scheduler SHOULD be smart enough to schedule the Task Futures on the same node that has the Partition Future. In #1136 I implemented that variant as well, and it works reasonably well there. It should be tested on a multi-node cluster, though, to make sure we really don't send the result of Partition Futures around

uellue, 2021-11-12:
A small side quest: The tqmd progress bar can't track the number of tasks anymore, but should track processed nav size vs. total nav size.

sk1p, 2021-11-12:
> A small side quest: The tqmd progress bar can't track the number of tasks anymore, but should track processed nav size vs. total nav size.

Yep, that's what I noticed when reading the code, too :)

uellue, 2021-11-12:
> As an alternative, one could fire off the reading process for a "raw partition" as a separate Future,

Hm, that would increase Dask overheads and wouldn't make much sense for other datasets with the current execution model. Probably best stick with the current implementation for now and worry about it later?

sk1p, 2021-11-12:
> I was also thinking about how to determine the starting partition size. If we start too large, we would "stare at an empty result in the GUI", and if we start too small we reduce max performance [...]

That's generally a good idea: when we want to change the size, we can create a "spread" of new partition sizes, in order to try different values concurrently.

> The results don't all arrive at the same time, meaning the main node can merge and send out new tasks without causing workers to wait.

If we want, one goal can be to decouple these two concerns: in `run_for_dataset_sync`, we are currently directly using `executor.run_tasks`. That could be replaced with a scheduling component which takes care of starting the next tasks as soon as one finishes, independently from / before the merging that is happening in `run_for_dataset_sync` itself.

uellue, 2021-11-12:
> That could be replaced with a scheduling component which takes care of starting the next tasks as soon as one finishes, independently from / before the merging that is happening in `run_for_dataset_sync` itself.

Sounds good! Task creation, merging and downstream result handling (GUI updates etc) can take a moment each, so it would be cool to do them concurrently resp. staggered. Making good use of hyperthreading, hopefully. :-)

uellue, 2021-11-12:
> With the tile scheme negotiation, we have the concept of a "base shape" - for partitioning, this could be handled in a similar manner.

👍 sounds good, ideally using the same code. :-)