LiberTEM/LiberTEM #387: Lossy and lossless data compression

uellue, 2019-07-02:
4D STEM data and other large-scale microscopy data would benefit significantly from efficient and effective compression. As an example, packed 12 bit integers for K2IS data take about 1/3 of the space compared to floats or int32 and are just as fast pixel-by-pixel with our fast unpacking routine. 

Series of diffraction and real-space images contain a lot of hidden redundancies and may contain information that is not relevant for subsequent analysis, such as noise. That means they are a good target for compression. This issue collects all ideas and discussion points regarding compression.

Idea collection:
* Test and compare classic compression algorithms. See no. 2 in https://libertem.github.io/LiberTEM/gsoc.html#project-ideas
* Video compression approaches
  * Lossless -- exploit redundancies between frames
    * How to handle data with more bit depth than usual video data? Codecs for camera raw data?
  * Lossy -- drop information that is not relevant
    * Bit depth, see above
    * Benchmark if/how lossy videocompression preserves information that is relevant for interpretation with computational methods vs information that is relevant for the human eye
    * Perform comparisons of processing raw vs compressed data -- blobfinder, radialfourier, virtual detectors etc
  * Hardware acceleration GPU/CPU? Some video codecs are highly optimized.
* Bit packing, dynamic bit depth
  * Exploit that frames contain a lot of low-intensity regions
* Autoencoders
  * Learn hidden lower dimensionality in typical data
  * How general can they be?
  * What is the correct quality function?
* Processing directly on reduced representation, similar to sparse matrix math?

Performance criteria:
* Decompression ideally just as fast as processing raw data from mass storage
* Reasonably fast compression to perform live during acquisition with high-performance setup. What is a good target?
* Good compression ratio that makes the effort worthwhile. 3x reduction like bit packing as baseline?

sk1p, 2019-08-23:
See also: https://www.usenix.org/system/files/atc19-gao.pdf - on file system compression, but interesting sections efficiency in general, like: memory usage, read amplification for ROI-like use-cases, ...

sk1p, 2020-01-22:
See also: https://arxiv.org/abs/1911.06013 (thx @robbmcleod)

never2average, 2020-02-24:
@uellue, will there be a screening process for a candidate's technical knowledge apart from the proposal.

uellue, 2020-02-25:
Hi @never2average, candidates prepare a pull request to fix a smaller issue before writing the proposal as part of the regular application process for projects under the umbrella of the Python software foundation. For details: https://libertem.github.io/LiberTEM/gsoc.html#writing-your-gsoc-application https://python-gsoc.org/index.html#apply

See https://github.com/LiberTEM/LiberTEM/issues?q=is%3Aissue+is%3Aopen+label%3A%22good+first+issue%22 for a list of issues that might be a good start.

This issue here would be suitable as a full project in GSoC.

sayandip18, 2020-02-29:
I would love to work on this issue for GSOC 2020. Will submitting 1 PR be enough for selection?

sk1p, 2020-03-01:
> Will submitting 1 PR be enough for selection?

Technically, yes. But as mentioned on [the Python GSoC page](https://python-gsoc.org/students.html#selectiontips): "Google intends this to be a way for new contributors to join the world of open source. The students most likely to be selected are those who are engaged with the community and hoping to continue their involvement for more than just a few months. It's more important to be a good community member than it is to be a good coder, for most projects!"

It is of course entirely your choice how to engage with our project - for some, this can mean working on multiple PRs to get to know the code base or to improve the documentation, for others, if could mean starting to write the proposal early, asking many questions, engaging in discussions in interesting issues, helping out others by answering questions in the chat etc.

sk1p, 2020-03-01:
Related paper: https://arxiv.org/pdf/1704.06976.pdf - the random access compression concept is similar to our ROI support, and the overall methodology could be adopted, modified for our specific requirements.

Btw: found while looking at GSoC proposals at CERN: https://hepsoftwarefoundation.org/gsoc/2020/proposal_ATLASMLcompression.html - which is also related to this issue!

sayandip18, 2020-03-16:
I have gone through the papers posted here. I only have a superficial idea of what's happening so far. I am afraid I don't have a clear roadmap of what needs to be done.

sk1p, 2020-03-17:
> I have gone through the papers posted here. I only have a superficial idea of what's happening so far. I am afraid I don't have a clear roadmap of what needs to be done.

The overall idea is: we want to store our data compressed, and there are many many options out there. We have specific requirements, mostly regarding performance. We would like to get some kind of overview which compression techniques are suitable for us, and possibly get one or more options implemented in our `ClusterDataSet`.

sayandip18, 2020-03-18:
Why `ClusterDataSet`? I thought that we need to write UDFs for the different compression techniques.

sk1p, 2020-03-18:
> Why `ClusterDataSet`? I thought that we need to write UDFs for the different compression techniques.

UDFs don't get to access the raw data. i.e. bytes as they are on the disk. If you have a look at [the UDF documentation](https://libertem.github.io/LiberTEM/udf.html), you will notice that a UDF only has access to array data, either as single frames or, in special cases, as tiles or partitions. 

But for compression, we need access to the raw bytes on the disk - that's where we want to save space! So this has to happen in a lower level, that of the `DataSet`s.

Also, we can't dictate what format our users save their data as - we can only improve our own data format, which currently is only the `ClusterDataSet`.

sayandip18, 2020-03-21:
How do we implement the FPGA based compression? Using Verilog?

sayandip18, 2020-03-21:
Another thing, so far I've been unable to figure out how we can use bit packing to compress image. Do you have any plan?

sk1p, 2020-03-21:
> How do we implement the FPGA based compression? Using Verilog?

The idea goes somewhat in the direction of the next-gen consoles that were very recently released (Xbox X and PS5), which both include compression hardware to achieve up to 2x the I/O of the included NVMe SSDs. I'm not sure it is wise to include FPGA-based compression in your project, unless you already have a lot of experience and a development kit available. Sadly we have neither FPGA hardware available, nor do we mentors have practical experience with FPGAs (correct me if I'm wrong, @uellue!)

> Another thing, so far I've been unable to figure out how we can use bit packing to compress image. Do you have any plan?

Did you have a look at how bitpacking works, and the general idea? If not, that's the first step.

To understand how it helps, have a look how raw and uncompressed images, in form of numpy arrays, are represented in memory (or saved to disk using `numpy.tofile`). You could load one of the HDF5 [sample datasets](https://libertem.github.io/LiberTEM/sample_datasets.html), and then have a look at the binary representation of the pixels. Try to answer the questions: 

- What is the datatype, i.e. how many bytes per pixel (bpp) does the array use?
- For a given pixel: how many bits of these bytes are **actually** used? (hint 1: what is the most significant bit that is 1? hint 2: have a look at `np.log2`)
- How does a histogram of "bits required to store the information" (from the previous question) look like? How many pixels of the data set use, for example, less than half of the available bpp?
- Why can't we use less bpp for this dataset?
- Bonus: answer the same questions for  random data (for example generated by `np.random.randint(0, high=2**16-1, size=65536, dtype='u2')`)

If you have answers to these questions, it should become clear how bit packing can help.

robbmcleod, 2020-03-21:
IMO I would stick to bit-shuffling over bit-packing. Shuffling is simply less effort and does not assume anything about the histogram of the underlying data. Here's the original implementation:

https://github.com/kiyo-masui/bitshuffle

which I believe is available as an HDF5 filter, somewhere. Also, I believe blosc has its own implementation, one of which uses AVX2:

https://github.com/Blosc/c-blosc/tree/master/blosc

AVX512 might have novel instructions that can do even more, although I have to admit I've not been super impressed with AVX512 thus far.

sk1p, 2020-03-23:
@robbmcleod thanks for mentioning the bitshuffle implementations!

> I would stick to bit-shuffling over bit-packing. Shuffling is simply less effort and does not assume anything about the histogram of the underlying data.

Point of the GSoC project is to do a survey, that is, compare these different methods and existing implementations, on typical real-world data. I would want to wait for the results of this survey before deciding on a single method.

uellue, 2020-03-23:
> Sadly we have neither FPGA hardware available, nor do we mentors have practical experience with FPGAs (correct me if I'm wrong, @uellue!)

Yes, that's true. I mentioned this since very, very fast and efficient compressors and decoders for sound and video are widespread as part of graphics units today. That's why you can have hour-long video conferences on a smartphone. Furthermore, hardware accelerators for, e.g., [gzip](https://www.ebay.de/itm/AHA-GZIP-AHA367PCIe-10-0-Gbits-sec-Compression-Decompression-Accelerator-Card-/401496720988) are available at reasonable cost.

That means a hardware-based solution is perhaps currently beyond our expertise, but can be an option in the future.

SahanWickramage, 2020-03-26:
I would like to work on this issue.

sk1p, 2020-03-28:
> I would like to work on this issue.

Hi @SahanWickramage - please note that it is very late in the application process, so if you plan to submit a proposal, please send us a draft ASAP! Also, please do thoroughly read https://python-gsoc.org/index.html#gettingstarted and https://google.github.io/gsocguides/student/ and let us know if you have any questions!

SahanWickramage, 2020-03-29:
Hello @sk1p. Thank you for consideration and sorry for the inconvenience. I sent you and @uellue an email including a link to my proposal. I got the email addresses from the LiberTEM GSoC page (https://libertem.github.io/LiberTEM/gsoc.html).

sayandip18, 2020-03-29:
@SahanWickramage  just to let you know, I am working on this issue too. You are free to choose any issue you want to work on, though. All the best for GSOC 2020!!!

uellue, 2022-07-28:
LiberTEM does support a range of compressed input formats already, we just follow what detector vendors and other software packages implement for the time being. Closing here in favor of discussions once we seriously start recording data and require compression.

closed 2022-07-28 by uellue